greatscottgadgets/ubertooth #414: Error when I tried to install ubertooth

nabiladn, 2020-09-04:
Excuse me. I want to ask something related with installing ubertooth. I followed all the tutorial from this link (https://wiki.elvis.science/index.php?title=Bluetooth_Sniffing_with_Ubertooth:_A_Step-by-step_guide). It was all working until the ubertooth tools installation step. I was expecting that the result is done 100%, but it said that there is something wrong with my library ubertooth. 

**root@kali:~/ubertooth-2018-12-R1/host/build# make**

make[1]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
make[2]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
Scanning dependencies of target ubertooth
make[2]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make[2]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
[  2%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o                                                                      
[  5%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_callback.c.o                                                             
[  8%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_control.c.o                                                              
[ 11%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth_fifo.c.o                                                                 
[ 14%] Linking C shared library libubertooth.so
/usr/bin/ld: CMakeFiles/ubertooth.dir/ubertooth_callback.c.o:(.bss+0x0): multiple definition of `packet_counter_max'; CMakeFiles/ubertooth.dir/ubertooth.c.o:(.bss+0x20): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/build.make:131: libubertooth/src/libubertooth.so.1.1] Error 1
make[2]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make[1]: *** [CMakeFiles/Makefile2:322: libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
make[1]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make: *** [Makefile:130: all] Error 2

**root@kali:~/ubertooth-2018-12-R1/host/build# sudo make install**
make[1]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
make[2]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
make[2]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make[2]: Entering directory '/root/ubertooth-2018-12-R1/host/build'
[  2%] Linking C shared library libubertooth.so
/usr/bin/ld: CMakeFiles/ubertooth.dir/ubertooth_callback.c.o:(.bss+0x0): multiple definition of `packet_counter_max'; CMakeFiles/ubertooth.dir/ubertooth.c.o:(.bss+0x20): first defined here
collect2: error: ld returned 1 exit status
make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/build.make:131: libubertooth/src/libubertooth.so.1.1] Error 1
make[2]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make[1]: *** [CMakeFiles/Makefile2:322: libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
make[1]: Leaving directory '/root/ubertooth-2018-12-R1/host/build'
make: *** [Makefile:130: all] Error 2

**root@kali:~/ubertooth-2018-12-R1/host/build# sudo ldconfig**

**root@kali:~/ubertooth-2018-12-R1/host/build#** 

=*=

I'm using Kali Linux 2019.4 in VirtualBox, and the hardware im using is ubertooth one. For the rest, I'm just following what's in the [tutorial ](https://wiki.elvis.science/index.php?title=Bluetooth_Sniffing_with_Ubertooth:_A_Step-by-step_guide)given.

Thank you so much!

tjhasan, 2020-09-19:
I had this same error. The problem is the "multiple definition of `packet_counter_max';" part.

Refer to [this change](https://github.com/greatscottgadgets/ubertooth/pull/409/commits/abc1d87631c71fb5bf3be4b81327c66292d852ca) in order to fix it.

In short, go to file ` host/libubertooth/src/ubertooth_callback.c` and change `unsigned int packet_counter_max;` to `extern unsigned int packet_counter_max;`

Then compile and run it again.

RduMarais, 2020-10-05:
worked for me, thanks !

mikeryan, 2020-12-05:
Fixed in abc1d87

closed 2020-12-05 by mikeryan