greatscottgadgets/ubertooth #415: btle sniffing: Unreassembled Packets and Control Opcode: Unknown

tjhasan, 2020-09-19:
### Steps to reproduce
1. Install ubertooth tools on Kali Linux as described in the Installation page
2. Run command `ubertooth-btle -f -c /tmp/pipe`
3. Wait for a CONNECT_REQ to occur

### Expected behaviour
I expect the connection to be followed and subsequent packet information to be shown.

### Actual behaviour
I do receive packets after the connection, but all of the packets are either "Control Opcode: Unknown" or "L2CAP Fragment[Unreassembled Packet]

Here is a screenshot of what appears on my screen: [image]
I have checked this multiple times and have gotten the same results. Any help would be greatly appreciated. 

### Version information
**Operating system**: 
Distributor ID: Kali
Description: Kali GNU/Linux Rolling
Release: 2020.3
Codename: kali-rolling

**Ubertooth tools version (ubertooth-rx -V):**
2018-12-R1

**libbtbb version:**
2018-06-R1

**Ubertooth firmware version (ubertooth-util -v):**
2018-12-R1

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Wireshark version: 3.2.6
gcc 10.2.0

mikeryan, 2020-12-01:
Those look like encrypted packets. What hardware are you trying to sniff?

tjhasan, 2020-12-05:
> 
> 
> Those look like encrypted packets. What hardware are you trying to sniff?

General btle hardware for the most part. I haven't seen any differences between devices. Here is a list of devices i've tried capturing so far:

-Apple Airpods
-Galaxy Buds
-Polar H7 Heart Monitor

mikeryan, 2020-12-05:
Airpods and Galaxy Buds are classic Bluetooth devices, and ubertooth-btle won't be able to help you with those. The H7 should work with ubertooth-btle, but it depends on whether the app you're using requires encryption or not.

Which app are you using, and is it on an iPhone, Android device, or something else?

tjhasan, 2020-12-05:
For the H7, I am using an Android phone with the app "Polar Bear"

mikeryan, 2020-12-22:
I have an H7 and Polar Beat running on an Android phone. It took several tries for me to capture the connection (as usual) but I can confirm that it is not using encryption. Please give it a few more tries and see if you're able to get plaintext data.

closed 2020-12-22 by tjhasan