greatscottgadgets/ubertooth #416: PCAP Time Discontinuities

ussjoin, 2020-09-28:
### Overview
I've been doing a lot of long Ubertooth captures recently, using Wireshark for analysis, and I've noticed that Ubertooth consistently reports extreme (appx 427 second) timing discontinuities. I've attached two PCAPs from the Virginia COVIDWISE applications (sending the Google/Apple Exposure Notification beacons; see, e.g., [the Google documentation on this](https://www.google.com/covid19/exposurenotifications/)) taken inside a Ramsey box (ensuring no other traffic would accidentally be caught). One was captured using an Android device as the source, the other with an iPhone as the source.

### Steps to reproduce
1. Create a FIFO: `mkfifo /tmp/pipe2`
2. Start a capture into the pipe: `/usr/local/bin/ubertooth-btle -n -c /tmp/pipe2`
3. Using Wireshark (or any PCAP tool), sniff from the FIFO and observe the timing irregularities. 

### Expected behaviour
There should not be 427-odd second jumps in "wall-clock time" as reported by Ubertooth.

### Actual behaviour
There were 427-odd second jumps in "wall-clock time" as reported by Ubertooth.

### Version information
**Operating system**:

MacOS 10.15.7

**Ubertooth tools version (ubertooth-rx -V):**

libubertooth 1.1 (git-19c69f2)

**libbtbb version:**

libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**

Firmware version: 2018-12-R1 (API:1.06)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

Confirmed both in Wireshark HEAD and in Wireshark 3.2.7.

### Output

See [TimingPCAPs.zip](https://github.com/greatscottgadgets/ubertooth/files/5294806/TimingPCAPs.zip). Extreme discontinuities occur at packets 435, 1595, 2035, 3179, 3609, 4771, 5211, 6365, 6789, 7938, 8368, 9532, 9964, 11122, 11550, 12701, and 13133. This makes it really difficult to do timing analysis, since the inter-packet arrival times are deeply corrupted (I had to export to Excel and do a ton of manual work). Note that the wall-clock times did *not* support the timing discontinuities---that is, packets would be arriving at a rate of 3-4 a second before, during, and after the discontinuities, as measured by me sitting there and watching.

### Potentially-Related Issues

It looks like other people have been reporting what *may* be a related issue for some time; see #202 and #251.

orgua, 2020-09-28:
could it also be ~429 seconds? my source-overview is more than 5 years away, but the time was derived from several sub-variables. it could be a 1-off when counting 100ns ticks with two separate 32bit registers

ussjoin, 2020-09-28:
[image]
[image]
You are correct---they're slightly more than 429 seconds, not 427. My apologies. Screenshots attached (showing the deltatime displayed column in Wireshark for the iPhone6S capture that I attached to the issue).

orgua, 2020-09-28:
there are some places in the firmware-code that do some edgy magic and still contain bugs. some are even shown / get obvious if you compile the file with gcc as cpp-code and include all warnings. but fixing them is hard, because of the lack of documentation telling something about the intention of the segments. 
i wish you best of luck!

orgua, 2020-09-29:
i didn't wanted it to sound like a rant. it was more like a starting point to find abnormalities and low hanging bugs. But fixing them needs often collab of one of the original devs and has to be in sync with libbtbb

Your error originates most likely in this routine (yes, there is a 100ns-counter): 
https://github.com/greatscottgadgets/ubertooth/blob/c2cc373bf9279ef3aab1177f34cf2e5be2137436/firmware/bluetooth_rxtx/bluetooth_rxtx.c#L731

And just to hint at some funky bits -> this whole file calculates mean-values of rssi-values that are done completely in dBm-Scale. Main code also subtracts some calibration-offsets directly... 
https://github.com/greatscottgadgets/ubertooth/blob/master/firmware/bluetooth_rxtx/ubertooth_rssi.c
https://github.com/greatscottgadgets/ubertooth/blob/c2cc373bf9279ef3aab1177f34cf2e5be2137436/firmware/bluetooth_rxtx/bluetooth_rxtx.c#L73

mikeryan, 2020-09-29:
Just to chime in here... thank you for your analysis and insights. The original devs are long gone, and a lot of the choices made over the years are questionable at best. I am more than open to major refactors and breaking changes if they make the project overall better. @orgua if you are on the fence about making any of these contributions, know that I would welcome them with open arms.

orgua, 2020-09-29:
I would love to do it. I even had parts of libbtbb and firmware modernized and simplyfied some years ago, I think there were even possible unreported bugs fixed. but: 
A: on big sections on the code I had questionmarks, so dominic-input or several days of research are needed for each
B: I don't have the time atm, or find the old code (3 days of work done, with at least 3 more weeks needed)
C: this needs testing-hardware, which I also don't have access to anymore
D: isn't the IC discontinued for several years now?

orgua, 2020-09-29:
and the routine where this related bug could be is needing lots of research. at least for me its cryptic :(

mikeryan, 2020-12-19:
This should be fixed in 9ecd365. Please reopen if you're still experiencing this issue.

closed 2020-12-19 by mikeryan

ussjoin, 2020-12-20:
@mikeryan Thank you so much! I confirm that in my testing with the current primary branch, I no longer experience this issue. Fantastic!