greatscottgadgets/ubertooth #417: Ubertooth installation on ubuntu 20.04

dianestephens, 2020-09-30:
### Steps to reproduce
1.sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \
pkg-config libpcap-dev python-numpy python-pyside python-qt4
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Unable to get prerequisites for Ubertooth installation.  Get the following messages:
unable to locate package python-pyside
unable to locate package python-qt4

### Version information
Ubuntu 20.04 in VirtualBox
(also same issue with Kali Linux 20)

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
unable to locate package python-pyside
unable to locate package python-qt4

```

omeround3, 2020-10-13:
I have the same issue with Kali Linux

mikeryan, 2020-12-01:
It looks like the dependencies for our spectrum analyzer are no longer available in Ubuntu 20.04. The build should proceed, you just will not get a graphical spectrum analyzer. The rest of the tools will work. Please give that a try and reopen this issue if you run into further issues.

closed 2020-12-01 by mikeryan

sysfu, 2023-02-25:
> ### Steps to reproduce
> 1.sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev pkg-config libpcap-dev python-numpy python-pyside python-qt4 2. 3.
>
> ### Expected behaviour
> Tell us [...]

This might be a viable workaround https://9to5answer.com/pyqt4-on-ubuntu-20-04-install