greatscottgadgets/ubertooth #419: Unable to access Ubertooth inside Docker

tranzmatt, 2020-10-23:
### Steps to reproduce
1. Install/build libtbb and ubertooth 2018-12-R1 on host OS (bionic) and inside Docker
2. Install kismet kismet-2020-04-R2 for the docker
3. Enable ubertooth-0 on kismet

### Expected behaviour

I want to see BT MACs on Kismet as detected by the Ubertooth

Expect same results from inside docker container

### Actual behaviour
Tell us what happens instead

Host:  $ 
ubertooth-util -V
ubertooth git-0adf33e (user@owk5) Fri Oct 23 11:48:40 EDT 2020

$ ubertooth-rx 
systime=1603470444 ch=53 LAP=9e8b33 err=0 clkn=1842 clk_offset=637 s=-63 n=-55 snr=-8
systime=1603470444 ch=53 LAP=9e8b33 err=2 clkn=1874 clk_offset=628 s=-63 n=-55 snr=-8
systime=1603470444 ch= 6 LAP=9e8b33 err=2 clkn=1999 clk_offset=3744 s=-75 n=-55 snr=-20
systime=1603470444 ch=55 LAP=9e8b33 err=0 clkn=2490 clk_offset=615 s=-65 n=-55 snr=-10
systime=1603470444 ch=57 LAP=9e8b33 err=1 clkn=3106 clk_offset=602 s=-65 n=-55 snr=-10

Enter docker: 
$ docker exec -it kismet_server_1 bash

root@fc0dc4b6107f:/src# ubertooth-rx -V
libubertooth 1.1 (git-0adf33e), libbtbb 1.0 (git-9d66440)

root@fc0dc4b6107f:/src# ubertooth-util -V
ubertooth git-0adf33e (user@owk5) Fri Oct 23 11:48:40 EDT 2020

** After enabling ubertooth-0 on kismet **

root@fc0dc4b6107f:/src# ubertooth-rx    
libUSB Error: No such device (it may have been disconnected):  (-4)
could not open Ubertooth device
ubertooth-rx - passive Classic Bluetooth discovery/decode

root@fc0dc4b6107f:/usr/local/bin# ubertooth-util -V
libUSB Error: No such device (it may have been disconnected):  (-4)
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One

### Version information
**Operating system**:
NVidia Jetpack 4.3.1 (Ubuntu Bionic 18.04)

**Ubertooth tools version (ubertooth-rx -V):**
$ ubertooth-rx -V
libubertooth 1.1 (git-0adf33e), libbtbb 1.0 (git-9d66440)

**libbtbb version:**
$ git status
HEAD detached at 2018-12-R1
nothing to commit, working tree clean

**Ubertooth firmware version (ubertooth-util -v):**
$ ubertooth-util -v
Firmware version: git-0adf33e (API:1.06)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

tranzmatt, 2020-10-23:
I build kismet on natively on the host OS and ubertooth is chirping away happily.  So the deeper issue seems to be libUSB is somehow mishandling it or perhaps something like udev?

mikeryan, 2020-12-01:
What is the output of lsusb from inside Docker?

straithe, 2021-01-26:
Do you still need assistance with this issue? We will close this issue if we don't get a reply within the next month.

straithe, 2021-02-25:
I'm going to close this as there hasn't been a response in a while, but please re-open this issue or open a new one if you still need assistance.

closed 2021-02-25 by straithe

tranzmatt, 2021-03-01:
Sorry, I missed this.  The issue seemed to be related to having udev system
visible to the system.  You need to pass /run/udev to the container.

Matt

straithe, 2021-03-01:
Thank you for the update!