greatscottgadgets/ubertooth #429: I get only error frames

sbehnsen, 2020-11-22:
I have recently purchased an ubertooth one. 
My original plan was to to sniff/analys bluetooth packets with Wireshark under windows. But it looks like this is not really supported...
So my next attempt was to use a Raspberry PI4. The ubertooth lib/tools exist as a ready package to install so it should be simple to use.
After updating the firmware to the lastest released version, ubertooth-rx did show some bluetooth traffic, but not really as expected. For example I have two devices communicating with bluetooth serial, but I don't ge to see two different LAPs...
loggin to a pcap file I can look at it with wireshark. However the logged data does seem to be generated when packets are send. But again the data is not what I expected. I can not really see any payloads and I also can not see the from/to bluetooth adresses :-( 
I then tried Kismet. Again installation was sraight forward, but Kismet just shows a pop up windows after a few seconds that is only detects error frames an for that reason it can not show anything...
(this confirms my suspision from the wireshark analysis that the captured data is unusable...)

The question now is what to try next? Any ideas what may be the cause of this?

mikeryan, 2020-12-01:
What version of the ubertooth tools are installed on the Pi? If it's not 2018-12R1 uninstall it and install that version. Make sure you have the latest firmware as well.

mikeryan, 2020-12-19:
Closing this due to inactivity. If you're still having this problem, please reopen the issue.

closed 2020-12-19 by mikeryan