greatscottgadgets/ubertooth #436: libUSB Error: Operation timed out:  (-7)

cdrdzsilva, 2021-01-04:
### Steps to reproduce
1. Just run kismet -c ubertooth
2. When login to Gui  http://localhost:2501
3. Im able to see the ubertooth devices but no activity show up and in the terminal start reporting a few errors...and the ubertooth one get disconnected and connected back.

### Expected behavior
I need to be able to run kismet for ubertooth.

### Actual behavior
Not able to properly install/ configure/ use this third app tool

### Version information
Im using VMWare Workstation 12 with VM Kali Linux 20.4

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth 2020-12-R1 (mikeryan@steel) Fri Dec 25 13:55:05 PST 2020

**libbtbb version:**
How i check this?

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2020-12-R1 (API:1.07)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Kismet....not sure how to check the version..i think i got one from 2019

### Output
KISMET - Point your browser to http://localhost:2501 (or the address of thiusb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)
usb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)
usb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)
usb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)
FATAL - Capture source did not get PING from Kismet for over 15 seconds; shutting down
ERROR: Data source 'ubertooth / ubertooth:39' ('ubertooth') encountered an 
       error: Pipe client closing - remote side closed pipe
ALERT: SOURCEERROR Source ubertooth has encountered an error. Kismet will 
       attempt to re-open the source in 5 seconds.  (3 failures)
ERROR: Source ubertooth has encountered an error. Kismet will attempt to 
       re-open the source in 5 seconds.  (3 failures)
INFO: Attempting to re-open source ubertooth
libUSB Error: Operation timed out:  (-7)
usb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)
usb_claim_interface error -6
libUSB Error: Operation timed out:  (-7)

Insert any commandline or build output here

root💀kali-IoT)-[~]
└─# lsusb            
Bus 002 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 018: ID 1d50:6002 OpenMoko, Inc. Ubertooth One
Bus 001 Device 003: ID 0e0f:0002 VMware, Inc. Virtual USB Hub
Bus 001 Device 002: ID 0e0f:0003 VMware, Inc. Virtual Mouse
Bus 001 Device 001: ID 1d6b:0001 Linux Foundation 1.1 root hub

mikeryan, 2021-01-04:
Are you able to use regular Ubertooth tools from inside the VM? Can you run `ubertooth-btle -n` and see if it displays BLE advertising packets?

cdrdzsilva, 2021-01-04:
Also,

I'm so confused about the whole installation...I was unable to find the proper step-by-step installation for this. For example...according to the wiki, I should install first the libbtbb stuff...then the ubertooth tool, and finally the third app tool... but there no details about which one should be integrated... For example i should put the lib directory inside the ubertooth file? and the kismet stuff??? i think i get all mess up...lol

More details below.

KISMET - Point your browser to http://localhost:2501 (or the address of thidrwx------  6 root    root      4096 Jan  4 18:40 .
drwxr-xr-x 19 root    root      4096 Jan  4 14:35 ..
-rw-r--r--  1 root    root      4503 Jan  4 14:32 .bashrc
drwx------  2 root    root      4096 Jan  4 16:40 .cache
-rw-r--r--  1 root    root     11656 Jan  4 14:35 .face
lrwxrwxrwx  1 root    root        11 Jan  4 14:35 .face.icon -> /root/.face
-rw-r--r--  1 root    root         0 Jan  4 16:41 .hushlogin
drwx------  2 root    root      4096 Jan  4 18:33 .kismet
-rw-r--r--  1 root    root     45056 Jan  4 18:15 Kismet-20210104-23-14-51-1.kismet
-rw-r--r--  1 root    root     81920 Jan  4 18:38 Kismet-20210104-23-35-43-1.kismet
-rw-r--r--  1 root    root     65536 Jan  4 18:40 Kismet-20210104-23-39-00-1.kismet
drwxrwxr-x 10 root    root      4096 Jan  4 16:48 libbtbb-2020-12-R1
-rw-r--r--  1 root    root    308434 Jan  4 16:46 libbtbb-2020-12-R1.tar.gz
-rw-r--r--  1 root    root       148 Nov  4 15:24 .profile
drwxr-xr-x  6 djchris djchris   4096 Jan  4 18:34 ubertooth-2020-12-R1
-rw-r--r--  1 root    root    763952 Dec 25 16:59 ubertooth-2020-12-R1.tar.

cdrdzsilva, 2021-01-04:
Hi mikeryan,

Yes i can run the following commands and worked ok...
ubertooth-btle -n
ubertooth-spec-ui ( I Can see the Graphics!!!)
ubertooth-rx

mikeryan, 2021-01-04:
It looks like your installation is correct and working, but I think there's an issue with how Kismet talks to the Ubertooth. It is probably using very aggressive USB timeouts, and VMware USB latency can be unpredictable and high. I think your best next move is to file a bug with Kismet.

cdrdzsilva, 2021-01-05:
well, I think I probably got a misconfiguration on my end for the kismet stuff....I found an old video from hack5 on youtube that explains how to properly install this stuff...but I think that doesn't match the current configuration. I also follow the following instructions but don't seem to be right for me...

sudo apt-get install libpcap0.8-dev libcap-dev pkg-config build-essential libnl-3-dev libncurses-dev libpcre3-dev libpcap-dev libcap-dev libnl-genl-3-dev
wget https://kismetwireless.net/code/kismet-2013-03-R1b.tar.xz
tar xf kismet-2013-03-R1b.tar.xz
cd kismet-2013-03-R1b
ln -s ../ubertooth-2015-10-R1/host/kismet/plugin-ubertooth .
./configure
make && make plugins
sudo make suidinstall
sudo make plugins-install
Add "pcapbtbb" to the "logtypes=..." line in kismet.conf   ( I NEVER FOUND THIS LINE ON THAT FILE). Neither is located under /usr/src/ were was recommended... i think new installation should be under /etc/kismet.conf... i think i need to troubleshoot kismet integration...with ubertooth....now that got that new GUI stuff...seems to be more user-friendly but unfortunately, nothing get working for me... not sure what is wrong or how I can troubleshoot these issues...i love the fact to get a finder where give the (Items counts) of Bluetooth devices around me..... like Mike Ossmann do in that interview back in 2011. https://www.youtube.com/watch?v=KSd_1FE6z4Y (minute 31)

mikeryan, 2021-01-05:
Those instructions you found are for legacy Kismet that is no longer supported by us or the original author. What instructions did you follow to install Kismet 2020-12-R3?

cdrdzsilva, 2021-01-05:
Well, Im driving crazy, lol.. that being said...i follow a few sites...instructions..... these are the links...not sure which one get the job done... probably none....

https://medium.com/@playswithfir3/kali-kismet-and-ubertooth-6452267a986
https://www.kismetwireless.net/docs/readme/datasources_bluetooth/
https://www.kismetwireless.net/docs/readme/quickstart/
https://www.kismetwireless.net/docs/readme/packages/
https://online-it.nu/kali-linux-2018-4-kismet/

Some of the commands run it..
 history
    3  sudo apt-get install cmake libusb-1.0-0-dev make gcc g++ libbluetooth-dev \pkg-config python3-numpy python3-qtpy
    4  wget https://github.com/greatscottgadgets/libbtbb/archive/2020-12-R1.tar.gz -O libbtbb-2020-12-R1.tar.gz
    5  tar -xf libbtbb-2020-12-R1.tar.gz
    6  cd libbtbb-2020-12-R1
    7  mkdir build
    8  cd build
    9  cmake ..
   10  make
   11  sudo make install
   12  sudo ldconfig
   13  cd
   14  wget https://github.com/greatscottgadgets/ubertooth/releases/download/2020-12-R1/ubertooth-2020-12-R1.tar.xz
   15  tar xf ubertooth-2020-12-R1.tar.xz
   16  cd ubertooth-2020-12-R1
   17  cd host
   18  mkdir build
   19  cd build
   20  cmake ..
   21  make
   22  sudo make install
   23  cd /ubertooth-2020-12-R1/ubertooth-one-firmware-bin
   24  cd ubertooth-2020-12-R1/
   25  cd
   26  cd /ubertooth-2020-12-R1/ubertooth-one-firmware-bin
   27  ls -la
   28  cd ubertooth-2020-12-R1
   29  ls -la
   30  cd ubertooth-one-firmware-bin
   31  ls -la
   32  ubertooth-util -v
   33  lsusb
   34  ubertooth-util -v
   35  sudo apt install build-essential git libmicrohttpd-dev pkg-config zlib1g-dev libnl-3-dev libnl-genl-3-dev libcap-dev libpcap-dev libnm-dev libdw-dev libsqlite3-dev libprotobuf-dev libprotobuf-c-dev protobuf-compiler protobuf-c-compiler libsensors4-dev libusb-1.0-0-dev python3 python3-setuptools python3-protobuf python3-requests python3-numpy python3-serial python3-usb python3-dev librtlsdr0 libubertooth-dev libbtbb-dev
   36  cd /tmp
   37  wget https://www.kismetwireless.net/code/kismet-2019-04-R1.tar.xz
   38  tar xf kismet-2019-04-R1.tar.xz
   39  cd kismet-2019-04-R1
   40  ln -s ../ubertooth/host/kismet/plugin-ubertooth ./
   41  KIS_SRC_DIR=/tmp/kismet-2019-04-R1
   42  ./configure
   43  make && make plugins
   44  history
   45  sudo apt install ubertooth
   46  apt remove kismet kismet-plugins
   47  wget -O - https://www.kismetwireless.net/repos/kismet-release.gpg.key | sudo apt-key add -
   48  echo 'deb https://www.kismetwireless.net/repos/apt/git/kali kali main' | sudo tee /etc/apt/sources.list.d/kismet.list
   49  apt update
   50  apt install kismet
   51  ubertooth-specan-ui

cdrdzsilva, 2021-01-05:
I think I was able to use it even with the bug of disconnecting/connecting loop... after running these commands....

219  kismet -v
  220  git clone https://www.kismetwireless.net/git/python-kismet-db.git
  221  git clone https://www.kismetwireless.net/git/python-kismet-rest.git
  222  git clone https://www.kismetwireless.net/git/python-kismet-external.git
  223  git clone https://www.kismetwireless.net/git/android-smarterwifi.git
  224  git clone https://www.kismetwireless.net/git/spectools.git
  225  git clone https://www.kismetwireless.net/git/android-pcap.git
  226  kismet
  227  exit
  228  history
  229  ubertooth-specan-ui
  230  ubertooth-btle -n
  231  lsusb
  232  ubertooth-util -v
  233  kismet -c ubertooth
  234  kismet -c ubertooth:channel=39
  235  sudo service networking restart
  236  ps -ef | grep kismet
  237  ps -ef | grep kismet*
  238  sudo rm -rfv /usr/local/bin/kismet* /usr/local/share/kismet* /usr/local/etc/kismet*
  239  kismet
  241  sudo apt remove kismet
  242  kismet

[image]
[image]
[image]
[image]
[image]

cdrdzsilva, 2021-01-05:
Thanks, everything seems to be working now.

closed 2021-01-05 by cdrdzsilva