greatscottgadgets/ubertooth #437: Wireshark with ubertooth not working Cant create the pipe

cdrdzsilva, 2021-01-05:
### Steps to reproduce
1. I follow the instructions for installation
2. Prepare the pipe
3. Error that stated This version of Wireshark does not save pipe settings.

### Expected behaviour
I need to grab the packets

### Actual behaviour
Nothing happens

### Version information
Kali 20.4 with Wireshark
[image]

# wireshark -v
Wireshark 3.4.0 (Git v3.4.0 packaged as 3.4.0-1)

Copyright 1998-2020 Gerald Combs <gerald@wireshark.org> and contributors.
License GPLv2+: GNU GPL version 2 or later <https://www.gnu.org/licenses/gpl-2.0.html>
This is free software; see the source for copying conditions. There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiled (64-bit) with Qt 5.15.1, with libpcap, with POSIX capabilities (Linux),
with libnl 3, with GLib 2.66.3, with zlib 1.2.11, with SMI 0.4.8, with c-ares
1.17.1, with Lua 5.2.4, with GnuTLS 3.6.15 and PKCS #11 support, with Gcrypt
1.8.7, with MIT Kerberos, with MaxMind DB resolver, with nghttp2 1.42.0, with
brotli, with LZ4, with Zstandard, with Snappy, with libxml2 2.9.10, with
QtMultimedia, without automatic updates, with SpeexDSP (using system library).

Running on Linux 5.9.0-kali1-amd64, with Intel(R) Core(TM) i5-7300U CPU @
2.60GHz (with SSE4.2), with 3910 MB of physical memory, with locale en_US.utf8,
with libpcap version 1.9.1 (with TPACKET_V3), with GnuTLS 3.6.15, with Gcrypt
1.8.6, with brotli 1.0.9, with zlib 1.2.11, binary plugins supported (0 loaded).

Built using gcc 10.2.0.

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth 2020-12-R1 (mikeryan@steel) Fri Dec 25 13:55:05 PST 2020

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2020-12-R1 (API:1.07)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
I follow the instructions from here... https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide
256  sudo apt-get install wireshark wireshark-dev libwireshark-dev cmake
  257  cd libbtbb-2018-12-R1/wireshark/plugins/btbb
  258  cd libbtbb-2020-12-R1/wireshark/plugins/btbb
  259  mkdir build
  260  cd build
  261  cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
  262  make
  263  sudo make install
  264  sudo apt-get install wireshark wireshark-dev libwireshark-dev cmake
  265  cd libbtbb-2020-12-R1/wireshark/plugins/btbredr
  266  cd ..
  267  ls -la
  268  cd ..
  269  ls -la
  270  cd btbredr
  271  mkdir build
  272  cd build
  273  cmake -DCMAKE_INSTALL_LIBDIR=/usr/lib/x86_64-linux-gnu/wireshark/libwireshark3/plugins ..
  274  make
  275  sudo make install
  276  mkfifo /tmp/pipe
  277  exit
  278  wireshark
  279  wireshark -v

```

mikeryan, 2021-01-05:
You need to click the `+` button and add the pipe you created in `/tmp`

cdrdzsilva, 2021-01-05:
Well, i was able to add it before. I meant the message below.... i got a few new screenshots....now the command appeart to work in the terminal but nothing showed in the wireshark app..... :(

Screenshot five i noticed is pointing out to eth0 instead of the usb ubertooth...not sure how to fix this....

thanks!
[image]
[image]
[image]
[image]
[image]
[image]
[image]

mikeryan, 2021-01-05:
Looks like this is a recurrence of #436. I will need to investigate whether the USB timeouts are too aggressive in libubertooth.

cdrdzsilva, 2021-01-05:
I think it's working now! I just was selecting the wrong interface. I just select in the bottom /tmp/pipe and start capturing after running the command in the terminal.... both sides are sniffing data!

[image]
[image]

cdrdzsilva, 2021-01-05:
> Looks like this is a recurrence of #436. I will need to investigate whether the USB timeouts are too aggressive in libubertooth.

Agree on this one. Not sure how to resolve this yet...still have to manually disconnect and connect back ubertooth to make eventually work....

closed 2021-01-05 by cdrdzsilva