greatscottgadgets/ubertooth #438: libusb timeouts may be too aggressive for VMware

mikeryan, 2021-01-05:
VMware's USB stack (at least VMware Fusion) has unpredictable and sometimes high latency. This can lead to the user error message:
```
libUSB Error: Operation timed out: (-7)
```
Adjust these timeouts or switch to a non-polled use of libusb to alleviate this issue.

cdrdzsilva, 2021-01-05:
Great that you open another issue. Let me know if you want to do some tests...I can share my screen using any screen sharing software and do some troubleshooting...also let me know if you need me to run any test. I'm available to dig some more over this issue since I'm currently experiencing this bug.

cdrdzsilva, 2021-01-11:
Nothing founded yet for this problem? keep happening all the time.

blacklabssecurity, 2021-02-17:
Any updates here... I'm receiving this error but also new to using the Ubertooth (could easily be user error). Trying to run ubertooth-btle -f <ADDR_to_Follow> and it immediately results in libUSB Error: Operation timed out: (-7)

Ubertooth Version: 2020-12-R1