greatscottgadgets/ubertooth #439: Ubertooth-rx command show same LAP address for differents equipments

cdrdzsilva, 2021-01-11:
### Steps to reproduce
1. Run ubertooth-rx command
2. Start powering on/off bluetooth on devices
3. get same LAP for 3 equipments (Note10, Iphone 7+ and iwatch Series 5)

### Expected behaviour
I was expecting to get differents LAP for each device. I need to be able to identify them.

### Actual behaviour
Show the same LAP

### Version information
Vmware with Kali Linux

**Ubertooth tools version (ubertooth-rx -V):**
ubertooth 2020-12-R1 (mikeryan@steel) Fri Dec 25 13:55:05 PST 2020

**libbtbb version:**
libbtbb-2020-12-R1

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2020-12-R1 (API:1.07)

If you are reporting a problem that involves third party software 
I will like to use Wireshark and I did. But still I don't know how to match the info obtained by sniferring devices.

### Output
```
systime=1610390244 freq=2402 addr=8e89bed6 delta_t=2.833 ms rssi=-81
02 22 ab fc 31 e1 e0 d8 1b ff 75 00 42 04 01 80 60 d8 e0 e1 31 fc ab da e0 e1 31 fc aa 01 00 00 00 00 00 00 fa 93 a7 
Advertising / AA 8e89bed6 (valid)/ 34 bytes
    Channel Index: 37
    Type:  ADV_NONCONN_IND
    AdvA:  d8:e0:e1:31:fc:ab (public)
    AdvData: 1b ff 75 00 42 04 01 80 60 d8 e0 e1 31 fc ab da e0 e1 31 fc aa 01 00 00 00 00 00 00
        Type ff (Manufacturer Specific Data)
           Company: Samsung Electronics Co. Ltd.
           Data: 42 04 01 80 60 d8 e0 e1 31 fc ab da e0 e1 31 fc aa 01 00 00 00 00 00 00

    Data:  ab fc 31 e1 e0 d8 1b ff 75 00 42 04 01 80 60 d8 e0 e1 31 fc ab da e0 e1 31 fc aa 01 00 00 00 00 00 00
    CRC:   fa 93 a7

systime=1610390245 freq=2402 addr=8e89bed6 delta_t=131.550 ms rssi=-42
40 18 8f 51 2f e6 59 5a 02 01 1a 02 0a 0c 0b ff 4c 00 10 06 42 1e 26 4c b6 d8 98 bc f2 
Advertising / AA 8e89bed6 (valid)/ 24 bytes
    Channel Index: 37
    Type:  ADV_IND
    AdvA:  5a:59:e6:2f:51:8f (random)
    AdvData: 02 01 1a 02 0a 0c 0b ff 4c 00 10 06 42 1e 26 4c b6 d8
        Type 01 (Flags)
           00011010
               LE General Discoverable Mode
               Simultaneous LE and BR/EDR to Same Device Capable (Controller)
               Simultaneous LE and BR/EDR to Same Device Capable (Host)

        Type 0a (Tx Power Level)
           12 dBm
        Type ff (Manufacturer Specific Data)
           Company: Apple, Inc.
           Data: 10 06 42 1e 26 4c b6 d8

    Data:  8f 51 2f e6 59 5a 02 01 1a 02 0a 0c 0b ff 4c 00 10 06 42 1e 26 4c b6 d8
    CRC:   98 bc f2

systime=1610390245 freq=2402 addr=8e89bed6 delta_t=19.700 ms rssi=-73
02 22 ab fc 31 e1 e0 d8 1b ff 75 00 42 04 01 80 60 d8 e0 e1 31 fc ab da e0 e1 31 fc aa 01 00 00 00 00 00 00 fa 93 a7 
Advertising / AA 8e89bed6 (valid)/ 34 bytes
    Channel Index: 37
    Type:  ADV_NONCONN_IND
    AdvA:  d8:e0:e1:31:fc:ab (public)
[... 194 lines cut ...]

    Data:  d1 3f fe c5 17 48 02 01 1a 02 0a 18 0a ff 4c 00 10 05 05 18 14 9f b0
    CRC:   45 31 19

systime=1610390245 freq=2402 addr=8e89bed6 delta_t=22.646 ms rssi=-87
40 25 ff 5f 3e f4 7a 5b 02 01 06 1b ff 4c 00 0c 0e 00 7b a0 b4 85 84 d2 1e f4 a6 c2 8e 72 eb 10 06 41 1d 23 cd cc 58 d5 e4 74 
Advertising / AA 8e89bed6 (valid)/ 37 bytes
    Channel Index: 37
    Type:  ADV_IND
    AdvA:  5b:7a:f4:3e:5f:ff (random)
    AdvData: 02 01 06 1b ff 4c 00 0c 0e 00 7b a0 b4 85 84 d2 1e f4 a6 c
        Type 01 (Flags)
           00000110
               LE General Discoverable Mode
               BR/EDR Not Supported

        Type ff (Manufacturer Specific Data)
           Company: Apple, Inc.
           Data: 0c 0e 00 7b a0 b4 85 84 d2 1e f4 a6 c2 8e 72 eb 10 06 

    Data:  ff 5f 3e f4 7a 5b 02 01 06 1b ff 4c 00 0c 0e 00 7b a0 b4 85 
    CRC:   d5 e4 74
![Screenshot_2021-01-11_10_03_08](https://user-images.githubusercontent.com/50721096/104229076-1f8eb700-5422-11eb-968a-c0ee76cee63c.png)

```

skinnyrad, 2021-02-21:
This occurs when BLE devices are advertising. They do so on one of 3 channels and use a LAP of 8e89bed6. However, if I'm not mistake the AdvA parameter in your output should give you a more accurate bluetooth hardware address.

mossmann, 2021-06-04:
Thank you for answering, @skinnyrad!

closed 2021-06-04 by mossmann