greatscottgadgets/ubertooth #440: Building Firmware on Fedora 33

stefangweichinger, 2021-01-13:
### Steps to reproduce
1. follow README on Fedora33
2. try to build firmware

### Expected behaviour
build works

### Actual behaviour
build fails ;-)

### Version information
**Fedora 33**

**Ubertooth tools version (ubertooth-rx -V):**
current git checkout with xmas release

Just a general question, without detailled errors at first:

It seems that the [Build Guide](https://github.com/greatscottgadgets/ubertooth/wiki/Build-Guide) does not fully list all the necessary dependencies for building on Fedora? When I look at your `.travis.yml` I think that the most development happens on Debian, I assume?

Yesterday a friend of mine tried to build on a Fedora 33 system and we had to install additional packages and even then failed to build once (it failed at a late point, linking the .elf file, as far as I understand).

Is that correct, or is there someone successfully using and building the code on F33?
Maybe only the Build Guide needs some love? ;-)

Thanks, Stefan

jkbenaim, 2021-01-17:
Hello @stefangweichinger . I have updated the build guide. It now builds the latest releases of btbb and the Ubertooth tools. I believe this will solve your problem. Please try it out and let me know.

Thanks!

stefangweichinger, 2021-01-17:
I thank you, @jkbenaim  ... and point my friend at the updated Build Guide to test things. Great!

stefangweichinger, 2021-01-18:
feedback @jkbenaim 

Set up a Fedora 33 VM, ran all updates and rebooted, then followed the Build Guide.

After that I hopped to [Firmware](https://github.com/greatscottgadgets/ubertooth/wiki/Firmware)

As far as I understand Fedora 33 needs these 2 packages (you want a PR?):

```
sudo dnf install arm-none-eabi-gcc-cs  arm-none-eabi-newlib
```

The make-run unfortunately fails still with the original error:

```
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: ego.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:38: multiple definition of `clkn_offset'; bluetooth_rxtx.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:38: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: ego.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:36: multiple definition of `last_hop'; bluetooth_rxtx.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:36: first defined here
/usr/lib/gcc/arm-none-eabi/10.2.0/../../../../arm-none-eabi/bin/ld: ego.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:35: multiple definition of `clkn'; bluetooth_rxtx.o:/home/sgw/ubertooth-2020-12-R1/firmware/bluetooth_rxtx/ubertooth_clock.h:35: first defined here
collect2: error: ld returned 1 exit status
make: *** [../common.mk:407: bluetooth_rxtx.elf] Fehler 1
```

Maybe I miss(ed) some step, maybe not. Can you spot the issue? Thanks.

jkbenaim, 2021-01-18:
@stefangweichinger , Thanks for checking it out. Looks like the normal Build Guide is working now, but you still can't build the firmware.

I've looked at the code in question and it looks like a bug in the firmware code. I am surprised and impressed that it ever worked at all. I will reach out to someone who knows this code well, and I'll see if we can make it work. This will take a while.

If you really need to get going right away, then it's possible that an ancient version of GCC and binutils might compile this garbage. After all, we know that it could be built correctly at some point in the past. I think GCC 4.x and binutils 2.23 would be your best bet. I'm not going to help you go down that road though... :smile: 

Hang in there and I'll see if I can poke the right people.

stefangweichinger, 2021-01-18:
@jkbenaim We don't need that right now, no. I just wanted to check if we miss something or if the reason is "upstream".
No hurry.

Yes, the normal build works for me. I can do tests in my VM if it's helpful.

straithe, 2021-02-18:
It looks like this issue has been solved so I will be closing it now.

closed 2021-02-18 by straithe