greatscottgadgets/ubertooth #441: About hardware version issues

BG7YWL, 2021-01-18:
I have two Ubertooth One. R161 and R218 versions, respectively.
[image]

I noticed that GSG previously sold the R238 version of the Ubertooth One, then switched it to the R161 version for some reason.

I checked the open source material on GitHub and saw that the latest version of the hardware is consistent with the R218 version, and why GSG uniformly sells the R161 version of Ubertooth.

Because GSG has been selling version R161 and developers are also using version R161 for development, I will always report errors when using version R218 hardware to capture packages.

I'm using the latest 2020-12-R1 firmware and I'm getting an "Error: attempt to read past end of buffer" error.

This error has been resolved in the 2018-06-R1 version by searching issues.https://github.com/greatscottgadgets/ubertooth/issues/270#issuecomment-400824574

So, I'd like to ask why GSG has been selling version R161 while GitHub's latest open source hardware version is version R218. And I don't know any information about the R161 version on GitHub.

zhaohuansi, 2021-01-21:
@mossmann

mossmann, 2021-01-23:
Great Scott Gadgets has never sold anything but r161 to date. Your Rev 218 board was manufactured by someone else. Are you having trouble with the GSG unit?

BG7YWL, 2021-01-24:
> Great Scott Gadgets has never sold anything but r161 to date. Your Rev 218 board was manufactured by someone else. Are you having trouble with the GSG unit?

Dear Mr. Mossmann, thank you for your reply.
I'm curious why GSG has been selling the R161 version of hardware while GitHub's latest open source version of hardware is R218.
I bought the hardware of R218 from Taobao a few years ago, and the vendor said that the equipment came from SparkFun. Through testing, I found that there were many bugs in the software operation, but there were no such problems in the version of R161.

straithe, 2021-02-18:
Where are you seeing that GitHub's latest open source version of hardware is R218?

BG7YWL, 2021-02-19:
I'm not sure about the latest hardware version of GitHub's open source, but it's the same as the R218 version that SparkFun makes public.
[image]

mossmann, 2021-03-02:
Long ago we used Subversion (svn) for revision control for project Ubertooth. I had it configured to automatically update the PCB silkscreen with the svn commit number (which was sequential). We finalized the PCB for production as of svn commit 161, so that number appears on the silkscreen of the units we manufactured. Over time the svn commit numbers went up from there, mostly because of software changes, so some gerber files (PCB fabrication outputs) and other documentation files were produced that had numbers higher than 161. Evidently we were at svn commit 218 when I gave documentation to SparkFun for their web site.

The manufacturer of your r218 unit probably used gerber files from the SparkFun site, but I am certain that SparkFun did not manufacture that unit. (SparkFun has always purchased Ubertooth One from Great Scott Gadgets.)

Eventually we switched from svn to git and migrated to GitHub. Since then we have not had automatic updates of the silkscreen (partly because git commit hashes are not sequential).

There have not been any significant changes to the Ubertooth One hardware design since r161, so we at Great Scott Gadgets continue to use r161 on the silkscreen of the units we produce.

Units with numbers higher than 161 should not be functionally different unless there was some error made by their manufacturer. If your r218 unit does not perform correctly, you can try updating the firmware. If it has the same firmware as your r161 unit and still does not work correctly, I suggest contacting the seller of the faulty unit.

closed 2021-03-02 by BG7YWL