greatscottgadgets/ubertooth #444: Controlling ubertooth-rx memory usage

ziXet, 2021-03-18:
### Steps to reproduce
1. Run "ubertooth-rx" without any extra parameter
2. use htop to monitor memory usage

### Expected behaviour
Why the memory usage should go to 64MB? Is there any internal buffering?

### Actual behaviour
The memory usage of ubertooth-rx goes to 64MB and stays there.

### Version information
Operating system: Ubuntu

ubertoth-rx -V 
`libubertooth 1.1 (2018-12-R1), libbtbb 1.0 (2018-06-R1)`

orgua, 2021-03-18:
i think thats based on the reversing of the hopping pattern - its done on the cpu. Nothing unusual

ziXet, 2021-03-18:
> i think thats based on the reversing of the hopping pattern - its done on the cpu. Nothing unusual

I don't really understand what you mean by "it's done on the CPU" !

orgua, 2021-03-18:
as you might know the ubertooth has at least one microcontroller. But it is not capable enough to  bluetooth on its own. that's all i wanted to imply. whats the trouble with the 64 MB? What is there to control? 

The problem with breaking into the bluetooth-communication is similar to a rainbow-table. it saves a lot of compute time, but costs storage.

ziXet, 2021-03-19:
> as you might know the ubertooth has at least one microcontroller. But it is not capable enough to bluetooth on its own. that's all i wanted to imply. whats the trouble with the 64 MB? What is there [...]
I think I know where the confusion is coming from!

I believe you didn't try to reproduce the situation.

The issue is that the 64MB memory is not consumed at the beginning of running the program. It start with 1MB and gradually increase to 64MB in 3-4 minutes and will stay in 64 for ever. This doesn't look like a normal memory usage profile to me.

miek, 2021-03-19:
64MB is the size of the rx packet FIFO:
https://github.com/greatscottgadgets/ubertooth/blob/d1feeccbd2fba266695a0052caab872ca5bcc263/host/libubertooth/src/ubertooth_fifo.h#L27-L34
the reason why it doesn't all show up immediately is that Linux will not actually allocate it all immediately, only as it is used.

ziXet, 2021-03-19:
> 64MB is the size of the rx packet FIFO:
> https://github.com/greatscottgadgets/ubertooth/blob/d1feeccbd2fba266695a0052caab872ca5bcc263/host/libubertooth/src/ubertooth_fifo.h#L27-L34
>
>
> the reason why [...]

That was the answer I was looking for! Thank you!

closed 2021-03-19 by miek