greatscottgadgets/ubertooth #448: Attempting to set Ubertooth One to channel 1 sets it to channel -134

atroche, 2021-03-27:
### Steps to reproduce
1. Plug in Ubertooth One
2. Run `ubertooth-util -c1`
3. Run `ubertooth-util -c`

### Expected behaviour

I expect the output to contain `(Bluetooth channel 39)`

### Actual behaviour

Current frequency: 2268 MHz (Bluetooth channel -134)

### Version information
**Operating system**: macOS 10.15.7 (Catalina)

**Ubertooth tools version (ubertooth-rx -V):** `2020-12-R1`

**libbtbb version:** `2018-06-R1`

**Ubertooth firmware version (ubertooth-util -v):** 2020-12-R1 (API:1.07)

atroche, 2021-03-27:
I see that the issue with the version number is probably just a result of https://github.com/greatscottgadgets/libbtbb/issues/56

straithe, 2022-07-15:
Thank you for the update and working to find a cause of the issue. I will look into this.