greatscottgadgets/ubertooth #450: Programming firmware on custom PCB results in no USB activity

supamas, 2021-04-15:
caveat - I'm not a super strong programmer :)
### Steps to reproduce
1. Blank LPC1758
2. Program via UART in ISP mode after reset, Flash Magic on windows to bluetooth_rx_only.hex, verified hex against programmed LPC1758 and came back as a match
3. Cycle power and observe USB data lines

Also, I tried using an arch linux machine with lpc21isp and saw different results output below.

### Expected behaviour
I was expecting some activity on the data lines after programming and a reset but am not seeing any activity at all.

### Actual behaviour
No behavior on the usb data lines.  In fact, it doesn't look like it's executing any code at all.

### Version information
- Windows 10 for Flash magic
- Arch Linux as a USB host

**Ubertooth tools version (ubertooth-rx -V):**
n/a I think
**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**
unknown because I can't talk to the chip to extract that - firmware was taken from the git repo 

If you are reporting a problem that involves third party software 
- flash magic
- lpc21isp

### Output
When using flash magic, flash compare comes back good.  

lpc21isp:
USB3 is my ftdi friend by adafruit.  When in ISP mode I can see that the auto-baud seems to catch and it's trying to program.  At the end when it goes to verify it comes back saying  "
```
./lpc21isp -verify -debug3 -hex bluetooth_rx_only.hex /dev/ttyUSB3 115200 14746
...
COMPARE_ERROR: Source and destination data not equal.
```

supamas, 2021-04-15:
Something to add:
I didn't realize I had populated an LPC1758 instead of the LPC1756 I see on the Ubertooth One board I have.  Do I have to compile the firmware for a specific chip version?  I noticed that the SRAM and Flash on the 1758 is twice the 1756.  Would that cause programming issues?

supamas, 2021-04-20:
To build on this...

So I noticed that the XTAL input was not getting 12 MHz from the CC2400 chip.  I double checked my schematic and noticed that the GUARD pin was tied to 1.8V 🤦‍♂️ 

I think I have been programming the chip correctly so I took off a capacitor and supplied an external 12 MHz expecting to see USB activity but I ended up not seeing anything.

pmcna, 2021-04-28:
Issue originated on my custom PCB.

I don't see an option for closing this but this can be closed.

closed 2021-04-28 by supamas