greatscottgadgets/ubertooth #451: Differences between LPC175x chips?

pmcna, 2021-04-26:
Is there a known method for programming firmware to the different variants in the LPC175x famly?

In reading here (https://os.mbed.com/questions/54231/Burning-hex-file-on-LPC1768-flash-memory/) it looks like there are slight changes that are required depending on the chip selected.  The Ubertooth-one that I received has an LPC1756 on it, but my custom PCB used an LPC1758.  I've been able to swap out the chip on the ubertooth-one board with an LPC1758 and program through ISP the bluetooth_rx_only.hex using the lpcisp library but the program doesn't stick after a reboot.

Any thoughts/suggestions?

Thanks!

orgua, 2021-04-26:
It lies some years in the past, but if i remember correctly there was a special fuse that had to be set prior to programming. And did you compile the firmware for that exact IC?

pmcna, 2021-04-26:
Special fuse?  Interesting, I have not seen something like that yet.  

I have not compiled the firmware for the exact IC yet though.  I extracted the firmware from the ubertooth-one using FlashMagic, then changed the chip on FlashMagic to 1758 and then downloaded the same firmware but that didn't seem to cut it.  I'll give it a run compiling for the exact IC

pmcna, 2021-04-26:
Ok, I think false alarm on this specific issue.  I downloaded the bootloader.bin and now it seems to come up after a reset and dfu mode works as expected.

closed 2021-04-28 by pmcna