greatscottgadgets/ubertooth #477: Ubertooth running on OSX with Kismet

sagfed, 2021-12-04:
### Steps to reproduce
1.install ubertooth-2020-12-R1
2.install kismet from git
https://www.kismetwireless.net/git/kismet.git
3. run kismet -c ubertooth

### Expected behaviour
To see BTLE list of scan of BTLE

### Actual behaviour
Only one or two BTLE appears from apple probably linked to my Imac 

### Version information
OSX:10.14.6

**Ubertooth tools version (ubertooth-rx -V):**
Libubertooth 1.1 (2020-12-R1)
**libbtbb version:**
Libbtbb 1.0 (2018-06-R1)
**Ubertooth firmware version (ubertooth-util -v):**
Libubertooht 1.1 (2020-12-R1), libbtbb (2018-06-R1) 

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
Screen capture, see here attached
[image]
[image]

Insert any commandline or build output here
sudo kismet -c ubertooth

straithe, 2021-12-04:
Have you opened this issue in the Kismet repository?

sagfed, 2021-12-04:
Yes under the #402, thank you

I was then able after recompilation to see the ubertooth device via kismet
and after its activation only one BTLE device can been visible and all the
other one are not shown …

Don’t know yet, if it is kismet issue or thé ubertooth  driver that I
cannot compile ?

The screen print are from Kismet however you could refer to Kismet/#402
ticket …

Thank you in advance,

Sagfed

straithe, 2021-12-04:
Thank you for the extra information!

Based on the report we got in #476 we are thinking this is a Kismet issue since Ubertooth functionality when not using Kismet is working fine.

sagfed, 2021-12-04:
Probably right however :

The only issue as pointed out in the Kismet topic, is that the driver for
Kismet ubertooth is provided by greatscottgadgets, if I have understood
correctly.

Therefore, this driver cannot compile due a legacy issue (from one library
packet-source.h not provided) and according some other forum discussion as
well. (See in that respect #288 in which it is stated that the
Ubertooth-Kismet is no longer built.)

On the top of this it seems that the Kismet version 2021 could work without
it, but TBC because in my case it doesn’t, you could see in the screen
capture that the related packet BTLE are scanning neither my iPhone
Bluetooth when I switch it on.

See the 2 screen capture …

So the main question is if it is due to the driver or to the kismet SW that
seems to have replaced this driver  ?
This is the answer I would like to know …

So, I am waiting also the analysis of the Kismet forum on this related
topic.

Thank you in advance,

Sagfed

straithe, 2022-07-15:
Can you please provide a link to the issue you opened with Kismet?

straithe, 2022-12-05:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2022-12-05 by straithe