greatscottgadgets/ubertooth #480: Trying to see if Ubertooth could handle this Bluetooth Classic use case

dv1, 2021-12-09:
I have an old gadget here that has a remote controller. It predates Bluetooth 4.0, so it uses Bluetooth Classic 2.1 instead of BLE. The remote controller is really badly designed, so I want to create my own. I know that it communicates to the device via RFCOMM.

What I want to do is to capture all of the RFCOMM data traveling between these two devices. I do know the PIN (it is a hardcoded PIN), so I am not talking about somehow cracking an encryption. With all of this, would Ubertooth be of help here? I know its Bluetooth Classic sniffer capabilities are limited, but perhaps it suffices for my use case.

mikeryan, 2021-12-09:
Check out [btbr host code](https://github.com/greatscottgadgets/ubertooth/tree/master/host/python/ubtbr) and [firmware](https://github.com/greatscottgadgets/ubertooth/tree/master/firmware/btbr). If Ubertooth is capable of sniffing the device, that would be the toolset that can do it. I have not used it very much, so I can't offer much support on that side.

Have you thought of trying to man-in-the-middle the connection using a Linux host running BlueZ advertising an RFCOMM service? Upon receiving connection from the controller, you could establish a connection to the target device and transport the data between the two, logging it.

dv1, 2021-12-12:
I considered that, yes. Unfortunately, it is rather complicated due to nontrivial SDP involvement. Nevertheless, I'll give it a try, thanks.

straithe, 2022-04-30:
@dv1 Do you still need assistance with this issue?

dv1, 2022-04-30:
Sorry forgot to close. No, I'll close this, thanks.

closed 2022-04-30 by dv1