greatscottgadgets/ubertooth #483: Multiple Ubertooths Cannot Operate in KIsmet

me-elliott, 2021-12-29:
### Steps to reproduce
1.Have two Ubertooth-1's connected directly to two USB ports
2. Have KIsmet installed
3.Have ubertooth installed and updated firmware on all Ubertooth-2 to be 2020-12-R1 (API:1.07)
4. Run KIsmet
5. Add Sources
6. Try to add multiple Ubertooth devices in KIsmet. 

### Expected behaviour
Kismet should take in multiple data sources, and both Ubertooth-1's should be collecting data

### Actual behaviour
UBertooth-0 , which ever enummerates as the first position, will be able to load the data into KIsmet, but Ubertooth-1 will not be able to. KIsmet shows the error that it is trying to re-open source. Error is below:

INFO: Attempting to re-open source ubertooth-1
multiple Ubertooth devices found! Use '-U' to specify device number
libUSB Error: Operation timed out:  (-7)

However, I thought there was a patch in the Ubertooth package under KIsmet to deal with multiple Ubertooth devices .

### Version information
**Operating system**: Ubuntu 20.04.3

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2020-12-R1), l
ibbtb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**
2020-12-R1 (API:1.07)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Ubuntu 20.04 Focal (amd64, arm)
### Output
```
Insert any commandline or build output here
Issue also exists in Ubertooth's specan-ui (ubertooth-specan-ui). When only one is plugged in, I can see the spectrum, but when I plug two in, nothing is displayed. The error shows as:

Could not open Ubertooth device
Failed to run:  ubertooth-specan -d - -l 2400 -u 2483 -U -1

straithe, 2022-07-15:
Have you brought the multiple device request to the Kismet folks? https://www.kismetwireless.net/

straithe, 2023-01-05:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2023-01-05 by straithe