greatscottgadgets/ubertooth #487: Firmware use interrupt EINT3 of the LPC1756, but this chip has only one external Interrupt EINT0.

mmackdev77, 2022-02-19:
in source file ubertooth_cs.c

void cs_trigger_enable(void)
{
	cs_trigger = 0;
	ISER0 = ISER0_ISE_EINT3;
	IO2IntClr = PIN_GIO6;      // Clear pending
	IO2IntEnF |= PIN_GIO6;     // Enable port 2.2 falling (CS active low)
}
The cs_trigger_enable function enable the external EINT3 of LPC17XX so the CC2400 carrier sense signal can interrupt the PCX17XX.

My ubertooth-one stick use a LPC1756 chip (175X family). On this chip there is only ONE external interrupt pin : EINT0.

So this function would not work. It could work with a LPC176X chip witch have 4 external interrupt EINT0:1:2:3.

Am I missing something ? Are there some ubertooth-one  sticks using LPC176X chip ?

mikeryan, 2022-02-19:
This code path, if not dead, is basically dormant and untested. I haven't checked the datasheet for the LPC175x but I'll take your word for it. Feel free to submit a PR to fix this!

straithe, 2023-01-05:
[Great Scott Gadgets has retired Ubertooth One](https://greatscottgadgets.com/2022/12-22-ubertooth-retirement/) and we have no plans to lead any hardware or software enhancements at this time. We will leave this issue open and will continue to monitor pull requests should anyone take on implementing this feature request themselves.