greatscottgadgets/ubertooth #489: Ubertooth One does not show in ioreg or lsusb on MacOS Monterrey; Green Leds show

rebelbot, 2022-03-14:
### Steps to reproduce
1. Plug device into mac. 
2. Run: **ioreg -p IOUSB -l -w 0 | grep Ubertooth**  or **lsusb**   
3. Notice that nothing regarding ubertooth shows up. Also notice that running DFU also doesn't work.

### Expected behaviour
Commands yeild a device on the system. I have tried multiple Macs.

### Actual behaviour
All other devices enumerate including hubs and internal devices but not this device.

### Version information
**Operating system**: Mac OS Monterrey

**Ubertooth tools version (ubertooth-rx -V):** r161-p11, Don't have more details from the board. Bought several years ago but opened just now.

**libbtbb version:** N/A

**Ubertooth firmware version (ubertooth-util -v):**??? whatever was flashed on it in ~2017 from factory.

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
$ ioreg -p IOUSB -l -w 0 | grep Ubertooth
$
```

rebelbot, 2022-03-17:
Handed off to @mikeryan today. I am still expecting that it will not enumerate on USB. Also found that I bought this back in 2016 (and only opened it now!).

husky-koglhof, 2022-03-23:
Hi, I've used these documentation: [https://codemuch.net/posts/bluetooth-hacking/](https://codemuch.net/posts/bluetooth-hacking/)
Switched to ignore case at ioreg command and it works!
`ioreg -p IOUSB -l -w 0 | grep -i ubertooth`

straithe, 2022-04-07:
@rebelbot are there any updates?

rebelbot, 2022-04-08:
Just got it back today. The HW is not faulty and the firmware was updated to recent version as opposed to 2015 time frame.

straithe, 2022-04-28:
May I close this issue now, or do you still need assistance?

rebelbot, 2022-04-29:
We can close this for now.

closed 2022-04-29 by rebelbot