greatscottgadgets/ubertooth #490: Upgrading firmware fail

tblake0709, 2022-03-15:
### Steps to reproduce
1.Prerequisites that need to be installed before building libbtbb and the Ubertooth tools on Ubuntu VM.
2. Installed and ran cmd from Build Guide libbtbb and ubertooth controls
3. Updating firmware for device. ubertooth-dfu -d bluetooth_rxtx.dfu -r

### Expected behaviour
Firmware updated to 2020-12-01

### Actual behaviour
Recieved this error: 
ubertooth-dfu -d bluetooth_rxtx.dfu -r
bluetooth_rxtx.dfu: No such file or directory

### Version information
**Operating system**:Ubuntu 20.04.4 LTS

**Ubertooth tools version (ubertooth-rx -V):**libubertooth 1.1 (2020-12-R1), 

**libbtbb version:**2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**2015-10-R1 (API:1.01)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2022-07-15:
Have you tried to do the update without going through a VM? It is possible that there are passthrough issues between the VM and the hardware.

straithe, 2022-08-15:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2022-08-15 by straithe