greatscottgadgets/ubertooth #491: Malformed Packet

Geranium1024, 2022-03-24:
1. Malformed Packet
[image]
2.Version information
Firmware version: 2020-12-R1 (API:1.07)
ubertooth 2020-12-R1 (mikeryan@steel) Fri Dec 25 13:55:05 PST 2020
3.Output
$ubertooth-btle -n   
[image]

How to solve this problem? 
Thanks !

ztroop, 2022-05-17:
I'm seeing this as well. Any suggestions on how to proceed?

straithe, 2022-07-15:
@Geranium1024 did @ztroop's suggestion work for you?

straithe, 2022-08-15:
I'm closing this as there hasn't been a response in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2022-08-15 by straithe