greatscottgadgets/ubertooth #492: No module named Qtpy for ubertooth-specan-ui macos

corecryptics, 2022-04-02:
### Steps to reproduce
1.  Brew Install ubertooth
2. Run uberooth-specan-ui
3. Found three versions of qtpy in various places, none linked to launch ubertooth-specan-ui

Tell us what you expect should happen

### Actual behaviour
Traceback (most recent call last):
  File "/Users/julianr/tools/ubertooth/ubertooth-2020-12-R1/host/python/specan_ui/./ubertooth-specan-ui", line 29, in <module>
    from qtpy import QtCore, QtGui, QtWidgets
ModuleNotFoundError: No module named 'qtpy'

### Version information
MacOS Big Sur

**Ubertooth tools version (ubertooth-rx -V): **libubertooth 1.1 (2020-12-R1)

**libbtbb version:** libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):** Firmware version: 2020-12-R1 (API:1.07)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
Traceback (most recent call last):
  File "/Users/julianr/tools/ubertooth/ubertooth-2020-12-R1/host/python/specan_ui/./ubertooth-specan-ui", line 29, in <module>
    from qtpy import QtCore, QtGui, QtWidgets
ModuleNotFoundError: No module named 'qtpy'

Insert any commandline or build output here
```

straithe, 2022-07-15:
Thank you for opening this issue. I do not have much experience with macOS or have appropriate hardware to recreate this issue. Please bear with me while I find someone from the community who can assist with this issue.

mr-slonn, 2022-07-15:
I have exactly the same problem in kali linux on a windows 10 virtual machine, at first I tried to do the instructions on the latest version of kali linux and failed, now I installed version 2020.4 and get the same error

straithe, 2022-07-15:
Thank you for the update! Since you have found a solution, I am closing this issue.

closed 2022-07-15 by straithe