greatscottgadgets/ubertooth #495: Executing ubertooth-specan-ui gives ModuleNotFoundError: No module named 'specan'

paulinaacostac, 2022-05-25:
### Steps to reproduce
1. Running on macOS
2. cd into /Users/iotlab/libbtbb-2020-12-R1/build/ubertooth-2020-12-R1/host/python/specan_ui
3. Executing ubertooth-specan-ui

### Expected behaviour
All the builds are completed so I should be able to execute the specs_ui and it should show the spectrum analyzer

### Actual behaviour
It shows the following:

```
File "/usr/local/bin/ubertooth-specan-ui", line 32, in <module>
from specan import Ubertooth
ModuleNotFoundError: No module named 'specan'
```

When I run `ls` command I get:

specan is present, its just not finding the module somehow.

CMakeLists.txt		setup.py		ubertooth-specan-ui
README			       setup.py.in
build			       specan

Why is it not able to find specan which is in the same folder where I'm running this?

Any help appreciated!

straithe, 2022-07-15:
Thank you for opening this issue. I do not have much experience with macOS or have appropriate hardware to recreate this issue. Please bear with me while I find someone from the community who can assist with this issue.

gilmarwsr, 2022-08-23:
I've the same problem... it's annoying but I could execute ubertooth-specan-ui successfully inside the source code folder (I've tried brew install ubertooth in MacOS, and I think I nave all dependencies installed).
So, you need to navigate to source code location:
 {whatever}/ubertooth/host/python/specan_ui
and run:
 python ubertooth-specan-ui

I'm using python3.10, all dependencies installed via pip.