greatscottgadgets/ubertooth #497: Unknow DFU:

chuzairps, 2022-06-29:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2022-07-15:
Can you please provide more detail? This issue does not explain what concerns you are experiencing.

chuzairps, 2022-07-25:
Hi Greatscottgadgets/Ubertooth
Sorry I could not reply back to you. I have initiated new Issue# 501. I am
facing issue of permissions at the moment. Please give me you precious
advice to resolve this issue as I need it to accomplish my thesis where I
want to use ubertooth for authentication purposes... I will be very
thankful to you for your help.

Thanks & regards

chuzairps