greatscottgadgets/ubertooth #500: Ubertooth not capturing pairing process

foliagecanine, 2022-07-18:
### Steps to reproduce
1. `mkfifo /tmp/ubertooth`
2. `wireshark -k -i /tmp/ubertooth &` 
3. In Wireshark, apply display filter `btle.data_header.length > 0 || btle.advertising_header.pdu_type == 0x05`
4. `ubertooth-btle -f -c /tmp/ubertooth`
OR `ubertooth-btle -f -A38 -c /tmp/ubertooth`
OR `ubertooth-btle -f -A39 -c /tmp/ubertooth`  
5. Pair two BLE-capable devices together.

### Expected behaviour
Packets with the defined filter show up in Wireshark, including CONNECT_REQ.

### Actual behaviour
Even trying all three advertising channels, no packets in the filter show up. Only the following show up:
ADV_IND, ADV_NONCONN_IND, SCAN_REQ, SCAN_RSP

### Version information
**Operating system**:
Kali 2022.2  
**Ubertooth tools version (ubertooth-rx -V):**
`libubertooth 1.1 (2020-12-R1)`  
**libbtbb version:**
`libbtbb 1.0 (2018.12.R1-1)`  
**Ubertooth firmware version (ubertooth-util -v):**
`2020-12-R1 (API:1.07)`  
**Wireshark version**
`Wireshak 3.6.6`

### Additional notes
Another thing I found strange is that not all of my Bluetooth devices show up. In fact, when just observing the advertising packets, only two laptops and another person's headphones are broadcasting their names.  
Additionally, there appears to be no difference between using the three advertising channels, as all three devices seem to broadcast on all three channels.
My problem seems rather similar to the last post in #113, but that post got no response.

foliagecanine, 2022-07-26:
I also tried this with an ESP-WROOM-32 running the BLE_server.ino example file (to ensure it wasn't trying to connect with classic BT instead).  
Good news is it did show up on the advertising channels, and when I set it to a specific channel, it didn't show up on the others.  
Bad news is I still have yet to capture the pairing process, even after forgetting the device and reconnecting several times.  

Any help would be appreciated, but I've only got a few days left with the Ubertooth before I will no longer have access to it (at which point I will probably close this issue).

straithe, 2022-08-08:
Have you run the capture a few times while attempting to capture the pairing process? It is normal to do multiple capture attempts before seeing a pairing signal.

foliagecanine, 2022-08-08:
Yes. I paired, unpaired and forgot, and repeated several times. The filter showed one or two malformed packets, but nothing beyond that.

straithe, 2022-08-15:
I would repeat a few dozen times at least. It does sometimes take a bit of effort to capture a pairing. Can you please let me know once you've tried to capture the pairing process at least a couple dozen more times?

foliagecanine, 2022-10-24:
Closing for now, as I do not have access to the device anymore. I'll reopen if/once I have tested it again.

closed 2022-10-24 by foliagecanine