greatscottgadgets/ubertooth #501: control message unsupported error: -9

chuzairps, 2022-07-24:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

chuzairps, 2022-07-24:
# ubertooth-dump | dd
usb_claim_interface error -6
ubertooth-dump - output a continuous stream of received bits
Usage:
	-h this help
	-b only dump received bitstream (GnuRadio style)
	-c classic modulation
	-l LE modulation
	-U<0-7> set ubertooth device to use
	-d filename

This program sends binary data to stdout.  You probably don't want to
run it from a terminal without redirecting the output.
0+1 records in
0+1 records out
349 bytes copied, 0.00365931 s, 95.4 kB/s

chuzairps, 2022-07-24:
─# ubertooth-dfu -r                      
usb_claim_interface error -6
Correct permissions to access Ubertooth?

willcode, 2022-07-25:
Likely permission error. Does the output of `id` include `plugdev`?

chuzairps, 2022-07-26:
> plugdev

Thanks @willcode  for your reply. I have added already. stats being shared. 

$ sudo adduser root plugdev
The user `root' is already a member of `plugdev'.
                                                                                
┌──(uzi㉿KKL)-[/]
└─$ sudo adduser uzi plugdev 
The user `uzi' is already a member of `plugdev'.
                                                                                
┌──(uzi㉿KKL)-[/]
└─$ sudo adduser $LOGNAME plugdev
The user `uzi' is already a member of `plugdev'.

chuzairps, 2022-07-26:
@willcode @straithe : Thanks for helping.
After changing the id and group from root to uzi for libbtbb-2020-12-R1 it is stopped giving error of usb_claim_Interface error:6
rwxrwxr-x  10 uzi  uzi    4096 Jul 17 19:34 libbtbb-2020-12-R1
But, still I am unable to upgrade firmware. 

# ubertooth-dfu -d bluetooth_rxtx.dfu -r
Checking firmware signature
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported
control message unsupported

chuzairps, 2022-07-26:
─# ubertooth-util -vV                    
Firmware version: 2018-12-R1 (API:1.00)
control message unsupported
error: -9

chuzairps, 2022-07-28:
Hi, there. Thanks for great community and help.
One more issue I'm observing auto detection of ubertooth one. Each time when i connect it does not detect automatically. I run command hcitool inq then its lights flash up. Does it be the issue not going into the DFU mode?

straithe, 2022-08-15:
I don't understand your question. Can you please rephrase?