greatscottgadgets/ubertooth #503: libubertooth not included in debian bookworm

dragorn, 2022-08-24:
Looks like debian bookworm isn't currently including libubertooth (or libubertooth-dev):

https://packages.debian.org/search?searchon=names&keywords=libubertooth

libbtbb is still present however:

https://packages.debian.org/search?suite=default&section=all&arch=any&searchon=names&keywords=libbtbb

straithe, 2022-12-02:
Thank you for the information. Is there something you would like us to change, or are you just passing along information?

dragorn, 2022-12-02:
We discussed it on discord and you asked me to open the ticket. I’m passing along information per request.

Mousketeer, 2023-02-19:
The window for bookworm is closing, I fear the entire ubertooth ecosystem might be left out.

straithe, 2023-02-19:
Do you know the deadline or who to talk to to get Ubertooth added?

paultag, 2023-02-19:
The window has been missed. At this point you'd need to talk to the release
managers.

https://release.debian.org/bullseye/freeze_policy.html#soft

miek, 2023-03-03:
It turns out that it did actually make it into bookworm in time, the maintainers backported that fix <https://metadata.ftp-master.debian.org/changelogs//main/u/ubertooth/ubertooth_2018.12.R1-5.1_changelog> :tada:

paultag, 2023-03-03:
Slight bit of pendantry, that was a non-maintainer upload, so `bunk@` was nice enough to fix it for Bookworm, not the Debian maintainer(s) - I should have checked before saying the window has been missed. Sorry for the noise. We can likely close this bug.

straithe, 2023-03-18:
Thank you for the update. I can see here that it made it in: https://packages.debian.org/bookworm/libubertooth-dev

closed 2023-03-18 by straithe