greatscottgadgets/ubertooth #504: Clasic/BR sniffing

skintigh, 2022-09-29:
2 questions:

The docs say this cannot sniff BR, then they say it can sniff BR, then it can sniff some, then ubertooth-btbr can, but not really. Can it?

More specifically: say I know 5 channels are used by the system under test, can I use an ubertooth to sniff and receive all data from those 5 channels? The channels used are very spread out, so I can't use a HackRF and aliasing.

straithe, 2022-12-02:
Can you tell me more about what you mean by "sniff"?

skintigh, 2022-12-12:
I wanted to be able to capture signals transmitted between a device and a USB BT dongle. Preferably I wanted the ability to act as a man in the middle, but at the very least I wanted to capture and then replay a transmission.

Initially I wanted this for Blue Tooth Classic, but later learned the device I was testing actually used only 5 or 6 BT channels and was possibly 2Mbps, but still seemed to be based on BT Classic hardware and not BLE. I attempted to use a HackRF, but the channels are far apart thus I can only capture 2 or maybe 3 at a time with aliasing on the HackRF.

There seemed to be a lot of options for BLE pen tests, but not BT Classic.

rhysperry111, 2023-11-17:
Did you make any progress on this? I'm in a similar situation

skintigh, 2023-11-25:
The HackRF only wants to tune 1 of the 79 channels. You can use aliasing to trick it into receiving several channels. I don't recall the exact number, but I think it was 20 or less under ideal conditions, so you'd still miss at least 75% of the data. The dongle I was sniffing only used a few channels but spaced them far apart so it might as well have been 79. I would have needed new hardware to sniff it, but instead I got a new job ;)

rhysperry111, 2023-11-25:
Understandable lol. It's very slightly possible with the `ubertooth-btbr` tool once you build the firmware that's needed, but will only capture data of a specified master+slave MAC if it also managed to listen for a handshake.

Just out of interest, since you mentioned other hardware, did you have anything in mind? Been looking around at quite a few research papers, and even they seem to be using the ubertooth.

skintigh, 2023-12-07:
Yeah, I think I was looking at ubertooth as well. There were lots of cheaper options for sniffing only the advertising channels of BLE, and for 2 or 3 orders of magnitude more money you can sniff all of them. For my application it seemed like ubertooth could capture my handful of fixed channels, but probably only after a lot of low-level hacking. I should poke around with that again... someday... when free time is a thing that exists again...