greatscottgadgets/ubertooth #506: Kail fail to update the firmware

YY-chen555, 2022-10-25:
### Steps to reproduce
1.Now my ubertooth version is 2017,but i want it to be newlest.
[image]

2.when i use `ubertooth-dfu -d bluetooth_rxtx.dfu -r`
[image]

When I execute this step in the vmware kail, I get an error. Every time I execute it, It switches to DFU mode, and then the USB display cannot be recognized. At the same time, the four red indicators from dark to light on the hardware are looping the marquee.

Hwo can i solve this.Thanks

straithe, 2022-10-27:
I suggest trying to update your Ubertooth without using a VM. Most VMs have USB pass through errors that can make upgrading difficult.

YY-chen555, 2022-10-28:
> 我建议尝试在不使用 VM 的情况下更新您的 Ubertooth。大多数 VM 都存在 USB 直通错误，这会使升级变得困难。
You are right.
I have done the upgrade via JTAG.
Thanks

straithe, 2022-11-01:
May I close this issue as complete now, @YY-chen555 ?

YY-chen555, 2022-11-02:
> yes and thank you!

closed 2022-11-02 by YY-chen555