greatscottgadgets/ubertooth #507: Ubertooth does not build on macOS 12.6.1, python 3.10.8

michelcrypt4d4mus, 2022-10-31:
### Steps to reproduce
Follow the [build from git](https://ubertooth.readthedocs.io/en/latest/building_from_git.html) instructions on macOS using homebrew:
1. `brew install` as per [this issue's](https://github.com/greatscottgadgets/ubertooth/issues/502) updated recommendation seems to work fine, though I do get a warning about `libpcap` linking (see below)
2. building `libbtbb` from source seems to work fine though there are some ominous warnings about requesting the python version `''`
3. Running `cmake` for `ubertooth` finds `libbtbb` ok (or seems to) but throws warning about `Could NOT find LIBBLUETOOTH (missing:  LIBBLUETOOTH_LIBRARIES LIBBLUETOOTH_INCLUDE_DIR)`. Unclear if it actually succeeded.
4. `make` fails for `ubertooth` on missing `libusb`:
    ```
    [  3%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
    In file included from /Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:30:
    In file included from /Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:25:
    /Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth_control.h:25:10: fatal error: 'libusb-1.0/libusb.h' 
    file not found
    #include <libusb-1.0/libusb.h>
             ^~~~~~~~~~~~~~~~~~~~~
    1 error generated.
    make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o] Error 1
    make[1]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
    make: *** [all] Error 2
    ```
5. also fails if i set `LDFLAGS` / `CPPFLAGS` as suggested by homebrew for the pcap library

## Expected behaviour
It should build

## Actual behaviour
See `4.` above.

## Version information
**Operating system**: macOS 12.6.1

**Python**: 3.10.8

**Ubertooth tools version (ubertooth-rx -V):** `2020-12-R1` (i think? just checked out of git today)

**libbtbb version:** `/usr/local/lib/libbtbb.1.0.dylib` so i guess 1.0?

**Ubertooth firmware version (ubertooth-util -v):** Didn't get that far yet.

# Output
## `brew install` as per the other issue
```
==> libpcap
libpcap is keg-only, which means it was not symlinked into /opt/homebrew,
because macOS already provides this software and installing another version in
parallel can cause all kinds of trouble.

If you need to have libpcap first in your PATH, run:
  echo 'export PATH="/opt/homebrew/opt/libpcap/bin:$PATH"' >> /Users/uzer/.bash_profile

For compilers to find libpcap you may need to set:
  export LDFLAGS="-L/opt/homebrew/opt/libpcap/lib"
  export CPPFLAGS="-I/opt/homebrew/opt/libpcap/include"
```

## Build `libtbb`
```
$ make
Consolidate compiler generated dependencies of target btbb
[ 88%] Built target btbb
[100%] Built target pcapdump
[100%] Built target btaptap
Install the project...
-- Install configuration: ""
-- Installing: /usr/local/lib/pkgconfig/libbtbb.pc
-- Installing: /usr/local/lib/libbtbb.1.0.dylib
-- Up-to-date: /usr/local/lib/libbtbb.1.dylib
-- Up-to-date: /usr/local/lib/libbtbb.dylib
-- Installing: /usr/local/include/btbb.h
/opt/homebrew/lib/python3.10/site-packages/setuptools/dist.py:544: UserWarning: The version specified ('') is an invalid version, this may not work as expected with newer versions of setuptools, pip, and PyPI. Please see PEP 440 for more details.
  warnings.warn(
[... 105 lines cut ...]

  MACOSX_RPATH is not specified for the following targets:

   ubertooth

This warning is for project developers.  Use -Wno-dev to suppress it.

-- Generating done
-- Build files have been written to: /Users/uzer/workspace/ubertooth/ubertooth/host/build
```

## (Without `cmake`, `make` also fails)
```
uzer@comput0r:~/workspace/ubertooth/ubertooth/host/build(master)$ make
[  3%] Building C object libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o
In file included from /Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth.c:30:
In file included from /Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth.h:25:
/Users/uzer/workspace/ubertooth/ubertooth/host/libubertooth/src/ubertooth_control.h:25:10: fatal error: 'libusb-1.0/libusb.h' file not found
#include <libusb-1.0/libusb.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/ubertooth.c.o] Error 1
make[1]: *** [libubertooth/src/CMakeFiles/ubertooth.dir/all] Error 2
make: *** [all] Error 2
```

michelcrypt4d4mus, 2022-10-31:
Perhaps worth noting that I definitely have `libusb.h`... in fact I have a couple:

### `/opt/homebrew/include` has one:
```
$ find /opt/homebrew/ -iname "*libusb*"
/opt/homebrew//include/libusb-1.0
/opt/homebrew//var/homebrew/linked/libusb
/opt/homebrew//Library/Taps/homebrew/homebrew-core/Formula/libusb.rb
/opt/homebrew//Library/Taps/homebrew/homebrew-core/Formula/libusbmuxd.rb
/opt/homebrew//Library/Taps/homebrew/homebrew-core/Formula/libusb-compat.rb
/opt/homebrew//lib/pkgconfig/libusb-1.0.pc
/opt/homebrew//lib/libusb-1.0.dylib
/opt/homebrew//lib/libusb-1.0.0.dylib
/opt/homebrew//lib/libusb-1.0.a
/opt/homebrew//opt/libusb
/opt/homebrew//Cellar/libusb
/opt/homebrew//Cellar/libusb/1.0.26/.brew/libusb.rb
/opt/homebrew//Cellar/libusb/1.0.26/include/libusb-1.0
/opt/homebrew//Cellar/libusb/1.0.26/include/libusb-1.0/libusb.h
/opt/homebrew//Cellar/libusb/1.0.26/lib/pkgconfig/libusb-1.0.pc
/opt/homebrew//Cellar/libusb/1.0.26/lib/libusb-1.0.dylib
/opt/homebrew//Cellar/libusb/1.0.26/lib/libusb-1.0.0.dylib
/opt/homebrew//Cellar/libusb/1.0.26/lib/libusb-1.0.a
/opt/homebrew//Cellar/libusb/1.0.26/share/libusb
/opt/homebrew//Cellar/libusb/1.0.26/share/libusb/examples/testlibusb.c
/opt/homebrew//share/libusb
```

### `/usr/local/MacGPG2/include` also has one:
```
$ find /usr/local -iname "*libusb*"
/usr/local/MacGPG2/include/libusb-1.0
/usr/local/MacGPG2/include/libusb-1.0/libusb.h
/usr/local/MacGPG2/lib/libusb-1.0.dylib
/usr/local/MacGPG2/lib/libusb-1.0.0.dylib
```

1. I'm wondering if there's some way the `MacGPG2` copy of `libusb.h` is in the way?
2. This seems like it should be fixable by setting some kind of env var when running `cmake`? `CPPFLAGS` or something? but alas i haven't actually written C++ in like, a very long time, and i've never had to deal w/`cmake`.

ihatecascardo, 2023-12-29:
I ran into the same problem building the ubertooth-* tools regarding libusb-1.0.0.  I edited /host/libubertooth/src/ubertooth_control.h file and provided an explicit path to my libusb.  This seemed to work and the make install finished.  Unfortunately when I tried to execute ubertooth-btle it errored with this 

`dyld[36380]: Library not loaded: libubertooth.1.dylib  Referenced from: <C557E36A-C2A9-3B94-9918-28ED42DA88AB> /usr/local/bin/ubertooth-btle Reason: tried: 'libubertooth.1.dylib' (no such file)....`