greatscottgadgets/ubertooth #509: How do I run ubertooth-btbr?

raysvmc, 2022-11-21:
### Steps to reproduce
1. Installed ubertooth-2020-12-R1

### Expected behaviour
Can't find instructions on how to install and run ubertooth-btbr. I understand that it is dfferent firmware and I found the firmware file in hosts/python/ubtbr

### Actual behaviour
The standard tools work fine, but I don't see any docs on btbr.

### Version information
Ubuntu 18.04.6

**Ubertooth tools version (ubertooth-rx -V):**
2020-12-R1
**libbtbb version:**
same
**Ubertooth firmware version (ubertooth-util -v):**
N/A

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2022-12-02:
Do you have a link to ubertooth-brbr?

raysvmc, 2022-12-02:
@straithe thanks for following up. I have ubertooth.btbr in the python folder, do you mean that? Maybe I just need to run setup.py, but I didn't know.

raysvmc, 2022-12-05:
@straithe thanks, yes had I read that already. I don't know how to get to the point where these commands can be issued. Do I need to install something? I tried running as ./ubertooth-btbr and saw the error 
ModuleNotFoundError: No module named 'usb1'
Apologies if this is a dumb question.

straithe, 2022-12-06:
Just to confirm, after installing the firmware you tried running the commands on https://github.com/greatscottgadgets/ubertooth/tree/master/host/python/ubtbr in a Python shell, yes?

raysvmc, 2022-12-06:
I just tried that, although I don't see how it could work since those commands aren't python files, and it doesn't work. I think I must be missing something obvious, some kind of installation I didn't finish. I tried running python setup.py and python3 setup.py but it requires arguments and I haven't studied the setup.py yet.

raysvmc, 2023-05-12:
@jsmif thanks for the advice. I tried the setup.py install then when I ran ubertooth-btbr I got an error 'No module named usb1'.
Haven't given this any thought for a while and need to get back into it and give it serious thought. Also need to upgrade from Ubuntu 18.04.
Thanks again!

straithe, 2023-10-11:
Hi @raysvmc are you still experiencing this issue?

rhysperry111, 2023-11-20:
Did you get anywhere with running `ubertooth-btbr`? I'm currently stuck with #520

straithe, 2023-11-28:
I'm closing this as there hasn't been a response from the original poster in over 30 days. Please re-open this issue or open a new one if you still need assistance.

closed 2023-11-28 by straithe

Leadisan, 2024-04-30:
> @straithe @mikeryan FWIW when I tried to run ubertooth-btbr on Ubuntu 22.04, I got the following error:
> 
> ```
> ubertooth-btbr --help
> Traceback (most recent call last):
>   File "/usr/local/bin/ubertooth-btbr", line 3, in <module>
>     from ubtbr.btctl import *
> ModuleNotFoundError: No module named 'ubtbr'
> ```
> 
> I then had to go to `ubertooth-2020-12-R1/host/python/ubtbr` and run `sudo python setup.py install` (which is not mentioned anywhere in the above cited documentation). @raysvmc try that first.
> 
> After that I could at least run the program:
> 
> ```
> ubertooth-btbr --help
> Usage: ubertooth-btbr OPTIONS
> Options:
>   -f freq_offset: Set MOD_OFFSET register
>   -b max_ac_errors: Max bit errors in Syncword detection
> ```
> 
> However, I would note that the version of the tool that's being run there is from `/usr/local/bin/ubertooth-btbr` according to `which`, since I think I did a make install at some point.
> 
> If I instead invoke the ./ubertooth-btbr in `ubertooth-2020-12-R1/host/python/ubtbr`, I get the below, which has an error and won't exit with ctrl-c, doesn't do anything if typing "inquiry", but which is maybe still sniffing some stuff according to the output? (But probably not, since if I restart it and initiate a BT classic connect with Braktooth and from bluetoothctl from the same machine nothing like the below L2CAP line shows up in the output.)
> 
> ```
> ./ubertooth-btbr 
> 18:36:50 | INFO | Rx thread started
> 18:36:50 | INFO | USB connected
> 18:36:50 | INFO | Send idle
> Exception in thread Thread-1 (_rx_thread_main):
> Traceback (most recent call last):
>   File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
>     self.run()
>   File "/usr/lib/python3.10/threading.py", line 953, in run
>     self._target(*self._args, **self._kwargs)
>   File "/home/user/Downloads/ubertooth-2020-12-R1/host/python/ubtbr/ubtbr/btctl.py", line 804, in _rx_thread_main
>     assert(t == b'S')
> AssertionError
> 18:36:51 | DEBUG | Set eir BBHdr(lt_addr=0, type=DM1, flags=), acl: ACL(llid=L2CAP_START, flow=1, len=11): 0a 09 55 62 65 72 74 6f 6f 74 68
> ^C^C
> ^C
> ```
> 
> (Note: I subsequently found that my system was in a weird state because I couldn't even do ubertooth-util -v without a USB error. I unplugged and re-plugged and made sure there were no background ubertooth processess still running. Then I could run `sudo ./ubertooth-btbr` and it would launch without error, but still not see any packets when I manually connected to nearby things.)

I encountered the same problem as you, have you solved it now?

raysvmc, 2024-04-30:
@Leadisan @straithe Apologies on my absence. on Ubuntu 22.04 I was able to run sudo python3 setup.py install in the ubtbr folder and can now run ubertooth-btbr. Maybe I hadn't tried this before? It seems obvious in hindsight. So far I can't get it to monitor a connection for some reason. Fortunately we haven't had a problem with our product connecting anyway so this wasn't a priority. Thanks!

Leadisan, 2024-05-01:
Thanks for the reply, but I'm still getting this after running setup.py.
It's like it's stuck.[image: image.png]Could you tell me how to solve this
problem?

raysvmc, 2024-05-01:
@Leadisan so supposedly we can run ubertooth-ubtbr while we're running ubertooth-btbr. I can't find any ubertooth-ubtbr and I'm thinking it might be a good idea to install everything from scratch when I have time.

Leadisan, 2024-05-01:
Thank you for your reply. I'm sure I installed Python 3, but unfortunately, sometimes I can accidentally enter the command page of Ubertooth-btbr normally, but there are no commands that work inside. I've tried many ways to solve the problem, but to no avail. The reason I want to use this command is actually to achieve the process of Ubertooth sending Bluetooth page data packets. I have an idea. Can I use a C language program to achieve this myself? And I wonder how to compile the program and run it on Ubertooth?发自我的iPhone------------------ Original ------------------From: raysvmc  ***@***.***>Date: Thu,May 2,2024 0:21 AMTo: greatscottgadgets/ubertooth ***@***.***>Cc: Leadisan ***@***.***>, Mention ***@***.***>Subject: Re: [greatscottgadgets/ubertooth] How do I run ubertooth-btbr? (Issue#509)
@Leadisan so supposedly we can run ubertooth-ubtbr while we're running ubertooth-btbr. I can't find any ubertooth-ubtbr and I'm thinking it might be a good idea to install everything from scratch when I have time.

[
{
"target": "https://github.com/greatscottgadgets/ubertooth/issues/509#issuecomment-2088712990",
"url": "https://github.com/greatscottgadgets/ubertooth/issues/509#issuecomment-2088712990",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]