greatscottgadgets/ubertooth #512: New Kali install kismet apt update

tmb2077, 2022-12-18:
### Steps to reproduce
1.kismet can't find plugun
2. Sudo apt update won't work
3.chang directory

### Expected behaviour
Run kismet for first time, and do an apt udate

### Actual behaviour
I just installed Kali on external hard drive, suspect paths are messed up, I run kismet it tries to pull from wrong file place and there is no kismet plugin

### Version information
Kali newest version

**Ubertooth tools version (ubertooth-rx -V):**

**libbtbb version:**

**Ubertooth firmware version (ubertooth-util -v):**

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

straithe, 2023-01-03:
Have you opened an issue with Kismet?

tmb2077, 2023-01-03:
Yes I figured it out, I was running Kali Persistence and I needed to do a full install on my external hard drive to get all the extra ins and outs.   Sent from Mail for Windows From: StraitheSent: Monday, January 2, 2023 9:26 PMTo: greatscottgadgets/ubertoothCc: tmb2077; AuthorSubject: Re: [greatscottgadgets/ubertooth] New Kali install kismet apt update (Issue #512) Have you opened an issue with Kismet?

straithe, 2023-01-03:
Thank you for the update. I will close this issue as resolved.