greatscottgadgets/ubertooth #518: libUSB Error: Access denied (insufficient permissions):  (-3) could not open Ubertooth device

wdyst, 2023-06-09:
### Steps to reproduce
1. Install libbtbb
```wget https://github.com/greatscottgadgets/libbtbb/archive/2020-12-R1.tar.gz -O libbtbb-2020-12-R1.tar.gz
tar -xf libbtbb-2020-12-R1.tar.gz
cd libbtbb-2020-12-R1
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```
2. Install ubertooth tools
```wget https://github.com/greatscottgadgets/ubertooth/releases/download/2020-12-R1/ubertooth-2020-12-R1.tar.xz
tar -xf ubertooth-2020-12-R1.tar.xz
cd ubertooth-2020-12-R1/host
mkdir build
cd build
cmake ..
make
sudo make install
sudo ldconfig
```
3. Install wireshark
4. Run any ubertooth-util, such as ubertooth-util -v or ubertooth-btle -f -c /tmp/pipe  

### Expected behaviour
Command should execute, for -v it should display ubertooth-util version. (Without sudo)

### Actual behaviour
#### ubertooth-util -v:
Without sudo:
`libUSB Error: Access denied (insufficient permissions):  (-3)
could not open Ubertooth device
ubertooth-util - command line utility for Ubertooth Zero and Ubertooth One`

With sudo:
`Firmware version: MP-2017-03-R2-V1 (API:1.02)`

#### ubertooth-btle -f -c /tmp/pipe  
Without sudo:
`libUSB Error: Access denied (insufficient permissions):  (-3)
could not open Ubertooth device`

With sudo:
`ubertooth-btle: lell_pcap_ppi_create_file: : Permission denied`
### Version information
**Operating system**:
Linux Asahi (aarch64)
6.3.0-asahi-7-1-edge-ARCH (64-bit)

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (git-3c70829)

**libbtbb version:**
libbtbb 1.0 (git-b92743c)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: MP-2017-03-R2-V1 (API:1.02)

If you are reporting a problem that involves third party software 
(Wireshark/Kismet/etc), please report the version here.
Wireshark:
Version 4.0.5 (Git v4.0.5 packaged as 4.0.5-1).

### Comments
I can provide any additional info as needed. I've gone through all previous issues and attempted troubleshooting with those posts but nothing has solved the issue yet. Wondering if there was something obvious I am missing here, setup error, etc. Thanks!

martinling, 2023-06-09:
You need to install some udev rules, to tell your system to make the device available to regular users. See [this section in the host README](https://github.com/greatscottgadgets/ubertooth/tree/master/host#privledge-reduction).

wdyst, 2023-06-10:
Solution process:
1. Run the command provided in the documentation on [this page.
](https://ubertooth.readthedocs.io/en/latest/getting_started.html)`$ echo 'ACTION=="add" BUS=="usb" SYSFS{idVendor}=="1d50" SYSFS{idProduct}=="6002" GROUP:="plugdev" MODE:="0660"' > /etc/udev/rules.d/99-ubertooth.rules`
2. Got 
`could not open Ubertooth device` when running `ubertooth-specan-ui`
Followed instructions #294 
3. Attempted to update firmware by following the documentation on [this page](https://ubertooth.readthedocs.io/en/latest/firmware.html).
by running `ubertooth-dfu -d bluetooth_rxtx.dfu -r`
5. Ran into "libUSB Error: Command Error: (-1"
Followed documentation again and ran
`make clean all && make
ubertooth-dfu -r -d bluetooth_rxtx/bluetooth_rxtx.dfu`
6. Reran `ubertooth-dfu -d bluetooth_rxtx.dfu -r`
7. Success.
------
Hope this helps someone lol, and thanks @martinling

closed 2023-06-10 by wdyst