greatscottgadgets/ubertooth #520: ubertooth-btbr hangs after start

AstroTV, 2023-08-25:
### Steps to reproduce
1. `cd Downloads/ubertooth-2020-12-R1/host/python/ubtbr`
2. `sudo python3 setup.py install`
3. `sudo python3 ubertooth-btbr`

### Expected behaviour
A promt should appear where the user can enter some commands

### Actual behaviour
The execution hangs after the following line:
`11:05:53 | DEBUG | Set eir BBHdr(lt_addr=0, type=DM1, flags=), acl: ACL(llid=L2CAP_START, flow=1, len=11): 0a 09 55 62 65 72 74 6f 6f 74 68`

### Version information
**Operating system**:
Linux 5.4.0-150-generic #167~18.04.1-Ubuntu SMP Wed May 24 00:51:42 UTC 2023 x86_64 x86_64 x86_64 GNU/Linux

**Ubertooth tools version (ubertooth-rx -V):** 
libubertooth 1.1 (2020-12-R1)

**libbtbb version:**
libbtbb 1.0 (2018-06-R1)

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2020-12-R1 (API:1.07)

### Output
```
sudo python3 ubertooth-btbr
11:05:52 | INFO | Rx thread started
11:05:52 | INFO | USB connected
11:05:52 | INFO | Send idle
11:05:53 | DEBUG | Set eir BBHdr(lt_addr=0, type=DM1, flags=), acl: ACL(llid=L2CAP_START, flow=1, len=11): 0a 09 55 62 65 72 74 6f 6f 74 68
```

I can not exit the hanging program with a Keyboard Interrupt. I have the unplug the Ubertooth to make it throw an exception and exit

m10653, 2023-10-03:
You need to make sure that you have loaded the correct firmware on the Ubertooth. You must upload the ubtbr firmware for that command to work.

rhysperry111, 2023-11-17:
Did you find any solutions to this?

rhysperry111, 2023-11-20:
Just worked this one out. You need to manually build the firmware yourself (it requires the arm toolchain to be installed), then flash the btbr firmware.

Leadisan, 2024-04-30:
> 刚刚解决了这个问题。您需要自己手动构建固件（需要安装 arm 工具链），然后刷新 btbr 固件。

Excuse me,could you tell me how you solved it?

00:05:45 | INFO | Rx thread started
00:05:45 | INFO | USB connected
00:05:45 | INFO | Send idle
Exception in thread Thread-1 (_rx_thread_main):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/home/cao/ubertooth-2020-12-R1/host/python/ubtbr/ubtbr/btctl.py", line 804, in _rx_thread_main
    assert(t == b'S')
AssertionError