greatscottgadgets/ubertooth #523: Where is the nine test points in the hardware?

sada45, 2024-01-10:
Hi!
I have brought a Ubertooth One for testing and I want to collect the I/Q sample from the ATEST1/2 pin of the CC2400 with an external ADC. I find there is a group of "test points" in the schematic, which includes nine points. But there is not indictaion on the hardware. 
If find nine contact points behind the CC2400 (red circle in the picture), is these are test points?
[image]

martinling, 2024-01-10:
No, that group of 9 are vias to the ground pad on the back of the CC2400 chip. The test points are the exposed circles without holes in them. Here's a screenshot with the names visible:

[image]

ATEST1 and ATEST2 are the two test points just below the start of the "EXPAND" text.

sada45, 2024-01-10:
Wow, thank you very much for the quick reply!

closed 2024-01-27 by straithe