greatscottgadgets/ubertooth #530: ubertooth-btle hangs after working for 5-15 mins

sakura-3, 2024-03-14:
### Steps to reproduce
1.nohup ubertooth-btle -n -c sniff.pcap &

### Expected behaviour
keep capturing until I kill the process

### Actual behaviour
firmware hangs after working for 5-15 mins,with process still alive but the RXLED off.

### Version information
**Operating system**:
host:Windows 11
VM:ubuntu 20.04

**Ubertooth tools version (ubertooth-rx -V):**
libubertooth 1.1 (2020-12-R1), libbtbb 1.0 (2018-06-R1)

**libbtbb version:**
1.0

**Ubertooth firmware version (ubertooth-util -v):**
Firmware version: 2020-12-R1 (API:1.07)

There is no output indicating that the capture has stopped.This problem seems to have been reported in https://github.com/greatscottgadgets/ubertooth/issues/326#issuecomment-440701847 and fixed in the commit 53f44b5

Lpd738, 2024-03-14:
I had similar issues when I got my Ubertooth One. It was on firmware version 2020-12-R1 and libbtbb was the 2018-12-R1 if I recall correctly. I tried like heck to build and install on a up to date Kali tablet but it failed with something about a version not reporting in the stack trace dump.  Well after about 8 hours of debugging I realized it was setuptools! So I lowered the version to 65.0 and boom. It installed libbtbb 2020 from source and everything is working now, including Mirage.  So sudo pip3 install setuptools=65.0, make and install per the instructions and the reinstall setuptools to the latest version.  Apt only has 2018 and lower and since it's 2024 I'm not thinking it'll make it into that. Hope this helps a few people!

sakura-3, 2024-03-15:
I followed this [build guide](https://ubertooth.readthedocs.io/en/latest/build_guide.html) from a raw vm.Seems that it has already installed libbtbb 2020-12-R1 from source.I'm not sure about the exact version since `ubertooth-rx -V` gives `libubertooth 1.1 (2020-12-R1), libbtbb 1.0 (2018-06-R1)`.

Meanwhile,following your guide,I execute `sudo pip3 install setuptools==65.0`,reinstall libbtbb and Ubertooth Tools after remove previous versions following [this](https://github.com/greatscottgadgets/libbtbb).And it seems that nothing has changed.

So how did you **make and install per the instructions**.I‘m a newbie in Ubertooth One.Any further guidance would be appreciated.

sakura-3, 2024-03-15:
The VM is exactly built from ubuntu-20.04.6-desktop-amd64.iso.As for the logs,I only maintain the packets captured by `ubertooth-btle -n -c sniff.pcap`.I'd like to share it if it's helpful.Besides,if you do need some logs,just teach me how to get them,like tell me the commands or give me a script.

Lpd738, 2024-03-15:
OK. That's plenty for me to debug/figure out. I'll work on it this weekend and report back in a few days. Hopefully this helps a lot of people!

Lpd738, 2024-03-15:
Oddly I can't get it to install from source and show the correct version with the -V... But it did install. Anyone else have any ideas?

sakura-3, 2024-03-15:
I have some ideas for this.This [issue](https://github.com/greatscottgadgets/libbtbb/issues/56) pointed a text error.Related fix has  not been included into the lastest release.

Maybe this indicates that my problem has already been solved.I will start sniffing again and observe how long it can work normally this time.

sakura-3, 2024-03-16:
The capture lasted more than a hour this time but still hung without any signal.I guess that's because it's early morning here and the traffic is quite light.

sreevatsank1999, 2025-08-26:
Do we have a solution to this problem?

closed 2025-09-09 by sakura-3