greatscottgadgets/ubertooth #531: Fails to build with current Pythons: setuptools.extern.packaging.version.InvalidVersion: Invalid version: ''

hhlari, 2024-09-20:
The same problem is with libbtbb. This has been working fine earlier, but now fails:

[ 97%] Generating build/timestamp
Traceback (most recent call last):
  File "/home/lisoant/hardware/ubertooth/host/build/python/specan_ui/setup.py", line 15, in <module>
    setup(
  File "/usr/lib/python3/dist-packages/setuptools/_distutils/core.py", line 146, in setup
    _setup_distribution = dist = klass(attrs)
                                 ^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 318, in __init__
    self.metadata.version = self._normalize_version(self.metadata.version)
                            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/setuptools/dist.py", line 354, in _normalize_version
    normalized = str(Version(version))
                     ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/setuptools/_vendor/packaging/version.py", line 200, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
setuptools.extern.packaging.version.InvalidVersion: Invalid version: ''
make[2]: *** [python/specan_ui/CMakeFiles/specan.dir/build.make:74: python/specan_ui/build/timestamp] Error 1
make[1]: *** [CMakeFiles/Makefile2:682: python/specan_ui/CMakeFiles/specan.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

TwoBottomBuns, 2024-10-09:
As a temporary workaround, you can just set the version yourself to something. I set the version in `host/build/python/specan_ui/setup.py` to 1.0, it doesn't really matter what it is as long as it's not an empty string. You may have to do this on another file as well.

TwoBottomBuns, 2024-12-20:
You can tell what files you need to add it to from the beginning of the Traceback:

```
[ 97%] Generating build/timestamp
Traceback (most recent call last):
File "/home/lisoant/hardware/ubertooth/host/build/python/specan_ui/setup.py", line 15, in setup(
```

This tells you that you have to add a version number to `host/build/python/specan_ui/setup.py`. When I did this, I put 0.8 but I don't believe there was any reason why I picked that version number specifically. This is what my file looks like:

```
from distutils.core import setup

setup(
    name        = "specan",
    description = "A tool for reading spectrum analyzer data from an Ubertooth device",
    author      = "Jared Boone, Michael Ossmann, Dominic Spill",
    url         = "https://github.com/greatscottgadgets/ubertooth/",
    license     = "GPL",
    version     = '0.8',
    package_dir = { '': '/home/braedon/aec/bluetooth/ubertooth-2020-12-R1/host/python/specan_ui' },
    packages    = ['specan'],
    classifiers=[
        'Development Status :: 5 - Beta',
        'Intended Audience :: Developers',
        'License :: OSI Approved :: GNU General Public License (GPL)',
        'Programming Language :: Python',
        'Operating System :: OS Independent',
    ],
)
```

After you do that and try to rebuild, you may get another error that's very similar but in a different file. Just do the same thing to find the other file and add a version into the setup.py file.