greatscottgadgets/ubertooth #541: AliExpress Ubertooth not recognized

a-blender, 2025-01-28:
Hi @dominicgs,

I'm a contractor in Colorado and my company ordered 6 AliExpress Ubertooths from [this vendor](https://www.aliexpress.us/item/3256806802284917.html?spm=a2g0o.productlist.main.9.7f2372323cffCn&algo_pvid=9f666ddc-0446-4b6b-9c62-1e731fea20e4&algo_exp_id=9f666ddc-0446-4b6b-9c62-1e731fea20e4-4&pdp_npi=4%40dis%21USD%2189.22%2153.53%21%21%2189.22%2153.53%21%402101c59517304198069146013eafa9%2112000038961816492%21sea%21US%210%21ABX&curPageLogUid=bpxdhgYJ4u0g&utparam-url=scene%3Asearch%7Cquery_from%3A&gatewayAdapt=glo2usa) a few weeks ago.

The devices appear to be dead. We tested one today and it
1. Does not show up on `lsusb` on linux 20.04
2. Does not show up in `dmesg` after connecting pins 1 and 3 as per the [Firmware Docs](https://ubertooth.readthedocs.io/en/latest/firmware.html#how-to-update-firmware). It shows a big green light and 2 yellow, 1 red tiny ones on the board.
3. Does not show up on Windows PC or Zidag utility (a functional Ubertooth we have is recognized)

Could you help with this issue? Is it defective or could we get it working in some other way?

a-blender, 2025-01-29:
We are aware of that. I've reached out to the vendor and thought posting here was worth a shot as well. Thank you!

Lalabrook, 2025-07-09:
Finally did you get any solution ?